mavlink/qgroundcontrol #4281: Fix thumb stick user model: Show indicator when active, send only when used

LorenzMeier, 2016-12-03:
@DonLakeFlyer Users are not aware that manual inputs are made always once they enabled thumb sticks once.

A first step would be to not send thumb inputs until the user puts both thumbs onto the screen. While the thumbs are active the screen should indicate that its making inputs. And we need to design something in so the vehicle can know wether the user is making an input (has his thumbs down) or not. We need that to adjust the behaviour of the vehicle properly.

LorenzMeier, 2016-12-05:
@DonLakeFlyer We can try to lock it out on the PX4 side (its now the fifth user or so reporting the issue) but I think the real fix needs to happen on the QGC side. It needs to differentiate between joystick in use and joystick not in use, it needs to *tell the user* and it needs to send the status to the vehicle.

If you have a remote control switched on you always somehow intend to use it. Its a whole different story for a joystick or even worse thumb sticks.

DonLakeFlyer, 2016-12-08:
There could be some sort of "Take control now" on/off thing. I don't think just putting your thumbs down is the way to do it. Regular joysticks have the same problem. As soon as you turn them on they start sending data.

closed 2026-10-05 by github-actions[bot]