mavlink/qgroundcontrol #4288: RC RSSI / Battery Flickering

tops4u, 2016-12-04:
When only connected to USB Power and no RC is used the RSSI and/or the Battery Symbol are Flickering (OS X - 3.0.2). Battery is flickering from dark grey to light gray and flipping between 0.0V and N/A. 

Minor Problem - just reporting.

DonLakeFlyer, 2016-12-04:
What firmware/vehicle?

tops4u, 2016-12-04:
@DonLakeFlyer It is an AUAV-X2 (FMUv2) with 1.5.1 RC3 when connected via USB without Battery and RC TX switched off (and to note that it is a Spectrum Receiver with 7 single channels, which is connected to a PPM Encoder).

DonLakeFlyer, 2016-12-04:
Great, thanks for the detail.

tops4u, 2016-12-16:
Today I observed that when it is flickering there are very short readings of 100% followed by 116%  before showing that there is no RSSI available (which is true). I can't tell you where the Values are coming from. It could as well be a strange behaviour of my PPM Encoder?

tops4u, 2017-04-02:
This is most probably related to this: https://github.com/PX4/Firmware/issues/6954

So I'm closing this here.

closed 2017-04-02 by tops4u