mavlink/qgroundcontrol #4290: Please add Throttle(%) to the "value widget"

AndKe, 2016-12-05:
Along with voltage and other values, throttle percentage is an important thing to monitor.

Multirotors:
If I fly an copter that normally flies about 30% throttle , and suddenly is ~50%  - I instantly know  that it has lost thrust on one motor/propeller  , redundancy is lost - and it should be landed ASAP.

Planes: 
If level cruise is about 50% throttle, and the plane tends to pitch up, fly at 70% , with no wind or other reason, then we know its a sign of icing..

Antiheavy, 2017-04-07:
I agree this would be valuable.

Antiheavy, 2019-04-29:
+1 for this request again.  This is useful for many use cases, including in telemetry log replay of problem flights.

DonLakeFlyer, 2019-04-30:
I just wondering if non-power users will have any idea what to do with this?

DonLakeFlyer, 2019-04-30:
> This is useful for many use cases, including in telemetry log replay of problem flights.

The data is there. You can turn it on before playing back the log.

DonLakeFlyer, 2019-04-30:
Oh, wait my bad. The problem here is that Throttle % is not available for use even if you add it manually.

Antiheavy, 2019-05-04:
> problem here is that Throttle % is not available for use

I see "throttle" in the MAVLink Inspector under VFR_HUD section.  Can that be added to the list of available "Values" to turn on/off?

Antiheavy, 2019-07-28:
Throttle % widget value exists now in QGC master: https://github.com/mavlink/qgroundcontrol/pull/7424

Good to close this issue?

AndKe, 2019-07-28:
Yes, thank you.

closed 2019-07-28 by AndKe