mavlink/qgroundcontrol #4293: Joystick Settings and MANUAL_CONTROL issues

larryyonge, 2016-12-06:
The MANUAL_CONTROL MAVLink message sends the joystick information from QGC to PX4 when PX4 is compiled and run as SITL.  There are several issues related to the QGC joystick settings and the joystick throttle value that appears in the MANUAL_CONTROL message.  This issue is base on QGC build cloned on November 1, 2016.

1)	The QGC joystick settings "Full down stick is zero throttle" and "Center stick is zero throttle" appear to behave incorrectly:
a.	Selecting "Center stick is zero throttle" results in 0-1000 throttle values in the MANUAL_CONTROL message where 0 = full back joystick throttle position, 1000 = full forward throttle position, and "linear" in between (see issue 2) below).  I believe it should be 0 for joystick position between full back and center, and then ramp linearly from 0 at center joystick throttle to 1000 at full forward joystick throttle position.
b.	Selecting "Full down stick is zero throttle" results in 500-1000 throttle values in the MANUAL_CONTROL message where 500 = full back joystick throttle position, 1000 = full forward throttle position).   It should be 0-1000.

2)	The range of throttle values (between 0 and 1000) in the MANUAL_CONTROL message appear to be non-linear.  With the joystick calibrated (I tried this several times), when the joystick throttle is mechanically in the center, and the slider in QGC (joystick settings) is also centered, (and with the problem described in 1) above not fixed), the throttle value in the MANUAL_CONTROL message is 577 (should be 500 if linear) for the "Center stick is zero throttle" setting, and 788 (should be 750 if linear) for the "Full down stick is zero throttle".

3)	If the "Center stick is zero throttle" and “Spring loaded throttle smoothing” are both selected, then the throttle value in the MANUAL_CONTROL message appears to be stuck at 1000 (never changes regardless of the joystick throttle position).

4)	Not sure if this is an issue or not:  QGC initially populates the MANUAL_CONTROL message with x=1000, y=-1000, z=1000, r=-1000 (not the actual joystick values) and continues to send these values in this message for some time (I only checked up to 30 seconds).  I assume some event causes QGC to send the actual joystick values (and it does), but not sure what that event is.  I would expect the actual joystick values would be used from the beginning.

DonLakeFlyer, 2016-12-07:
@rjehangir Is there anything wrong here?

rjehangir, 2016-12-07:
@larryyonge I just checked the output of MANUAL_CONTROL from the most recent code and it appears to behave exactly as it should, I think. I checked out the code today, but I don't think any of this has changed since Nov 1.

1. When set to "Full down stick is zero throttle", I get 0 to 1000 from full down to full up on the joystick. For "Center stick is zero throttle", I get 0 - 0 from full down to center, and 0-1000 from center to full up.

2. The range is linear (as long as your joystick has a linear range). Note also that some joysticks, particularly the Xbox 360/One controllers, don't really have much deadband, so it's very hard to completely center the joysticks.

3. The spring throttle seems to work fine for me. If I move the stick down, the value decreases, if I move it up, the value increases. When center, the value remains the same.

4. I don't see this behavior, either. The joystick seems to send the correct values immediately.

Can you tell me more about how you are testing this? What type of joystick?

-Rusty

larryyonge, 2016-12-08:
Perhaps this has something to do with the joystick I am using:  Thrustmaster T-Flight Hotas X

I cloned the latest version of QGC this morning and repeated the previously mentioned tests, and got the same results at before.  I’m running Ubuntu 16.04 and QT Creator.  Here are the details from running the test this morning:

Running PX4 code SITL with iris and jMAVSim (still using Nov 1 build), joystick calibrated in QGC (QGC version cloned today), joystick physically centered (throttle detent) corresponds to center slider position in QGC.

Comments:
Sim flies correctly if joystick setting is "Center stick is zero throttle"
Sim will not arm if joystick setting is "Full down stick is zero throttle" since throttle is never zero (see test results):
  "WARN  [commander] Arming DENIED. Manual throttle non-zero."

Breakpoint set during run in QGC at line 1382, file UAS.cc, function UAS::setExternalControlSetpoint, then inspecting value of “thrust” and “newThrustCommand”

1381            // Send the MANUAL_COMMAND message
1382            mavlink_msg_manual_control_pack_chan(mavlink->getSystemId(),
1383                                                 mavlink->getComponentId(),
1384                                                 _vehicle->priorityLink()->mavlinkChannel(),
1385                                                 &message,
1386                                                 this->uasId,
1387                                                 newPitchCommand, newRollCommand, newThrustCommand, newYawCommand, buttons);

Each test is run with 3 joystick positions:
joystick full back & slider indicator full left / joystick centered & slider indicator centered / joystick full forward & slider indicator full right

QGC joystick setting – “center stick is zero throttle”
  thrust = 0 / 0.577338517 / 1    (back / center / forward)
  newThrustCommand = 0 / 577.338501 / 1000    (back / center / forward)

QGC joystick setting – “Center stick is zero throttle + Spring loaded  throttle smoothing”
  thrust = 1 / 1 / 1
  newThrustCommand = 1000 / 1000 / 1000

QGC joystick setting – “Full down stick is zero throttle”
  thrust = 0.5 / 0.788669229 / 1
  newThrustCommand = 500 / 788.66925 / 1000
  
Summary:
1) “thrust” range is 0.0-1.0 for "Center stick is zero throttle"
2) “thrust” range is stuck at 1.0 for "Center stick is zero throttle + Spring loaded  throttle smoothing"
3) “thrust” range is 0.5-1.0 for "Full down stick is zero throttle"

- Larry

larryyonge, 2016-12-08:
Regarding original item 4, values are updated in the MANUAL_CONTROL message once the joystick is moved.  Until the joystick is moved, the values are "x=1000, y=-1000, z=1000, r=-1000" as described above.  I'm guessing this is the intended behavior (joystick values are not populated in the MANUAL_CONTROL message until joystick movement is detected).

AndKe, 2018-08-24:
I flew with RealFlight Interlink controller today:  (QGC version is current master)
Just a note: throttle works as expected in MavProxy.
The joystick data for throttle goes from ~ -20000 ,to zero (center) then up to 20000. (verified in console)

In QGC, no matter how many times I calibrate it, carefully following on-screen instructions, the verification dot on the slider is not behaving properly, and flight is equally crazy:
To hover, the physical throttle stick position must be 80% or 90% up, 100 %.
or to make it clear:
to make the dot that slides in the calibration screen to be in the middle for throttle, the physical stick needs to be 80-90% up.

DonLakeFlyer, 2018-08-24:
> The joystick data for throttle goes from ~ -20000 ,to zero (center) then up to 20000. (verified in console)

@AndKe Verified using the Joystick log output?

AndKe, 2018-08-24:
@DonLakeFlyer `Is the joystick set to "Center stick is zero throttle"?`

... and you nailed it. shame on me,... I have been flying shakedown on a large (OEW = 23Kg)  OctoQuad all day .. then started to test joystick on a smaller hexa..   what can I say, sorry . I were simply bling to that option.

DonLakeFlyer, 2018-08-24:
It's a bad default in reality. I thought I had changed it. But I'll double check.

jaxxzer, 2019-01-28:
Is it ok to close this?

DonLakeFlyer, 2019-01-31:
I believe so

closed 2019-01-31 by DonLakeFlyer