mavlink/qgroundcontrol #4303: apt-get install gstreamer1.0* is not working on 16.04

uavcam, 2016-12-08:
On a ubuntu 16.04 64bit system the apt-get gstreamer1.0* will fail because there is a i386 dependency somewhere. and even after installing all the packages by hand the libgstreamer-plugins* dev packages for the gstreamer-video-1.0 are missing

PKGCONFIG   += gstreamer-1.0  gstreamer-video-1.0

this needs libgstreamer-plugins* packages. do not know which one exactly - you guys will figure it out. will be one of the dev packages.

dogmaphobic, 2016-12-08:
Yeah, I guess it's time to upgrade GStreamer to something newer. QGC will only support 1.5.2 across all platforms. I don't remember exactly which GStreamer version, but at some point the interfaces changed and it will require some work to get all platforms up-to-date. That also means you will no longer be able to "apt-get" on Ubuntu < 16.04.

uavcam, 2016-12-09:
i think the only way is to get rid of the problem is to supply the libs needed by the application and not a generic apt-get on all things that start with gstreamer1.0*

dogmaphobic, 2016-12-09:
Ubuntu is the only one giving headaches because it installs at the system level. All other platforms allow you to pull a specific version and have it isolated. I downloaded the current, stable Mac OS version (1.10.2) and it required no changes. It all just built and run. Ubuntu 16.04, even with trying to fix all of the above is failing to build, complaining the lack of packages, header files, the works. This will require quite a bit of work to sort it out.

If we are to include the libraries/headers, it would be Ubuntu 16.04 only and even that would cause a huge increase in the size of the repository. I'm not too thrilled with that idea and frankly, I don't have that kind of time for it right now.

DonLakeFlyer, 2016-12-09:
What is the state of Ubuntu version is 16.04 the new LTS?

uavcam, 2016-12-09:
ubuntu 16.04 is the new LTS, yes.

[qgcTesting] user [~]$ gst-launch-1.0 --version
gst-launch-1.0 version 1.8.2
GStreamer 1.8.2
https://launchpad.net/distros/ubuntu/+source/gstreamer1.0
[qgcTesting] user [~]$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
[qgcTesting] user [~]$ 

version is 1.8.2

i think its the same problem on most of the major linux distributions installing applications/libs via a packet systems. yum, apt, etc. will download a version out of your control.

> If we are to include the libraries/headers, it would be Ubuntu 16.04 only and even that would cause a huge increase in the size of the repository. I'm not too thrilled with that idea and frankly, I don't have that kind of time for it right now.

i do not think that will work because the gstreamer has its own way of loading plugins. its not loading shared libs via ld i think. static linking is not the way to either.

if you can give me a bunch of ubuntu version you want to support i will figure out for you what the nesseccary commandlines are to install the needed gst libs/binaries. i would suppose to support the last two (perhaps three) LTS versions, the support for 12.04 should run out next year. i can do that for debian too. at least jessie and wheezy. perhaps arch linux too - but no clue about the versions.

long things short: give me list of versions/distributions you want to support and i will get you the minimal command lines to install the needed gstreamer things.

dogmaphobic, 2016-12-09:
We use a tiny subset of GStreamer. For Android I just link what we need statically. I used "Ubuntu" above mostly as a synonym for Linux, given that Ubuntu is the only distro we officially support. For 14.04, pkgconfig took care of it. Something changed in 16.04.

uavcam, 2016-12-11:
pkgconfig is taking care of it in ubuntu 16.04 too. but the package struckture seems to be changed

yzzir, 2016-12-15:
I successfully built QGC today on Kubuntu 16.04 with streamingvideo and uvc support (qt.io Qt 5.5.1, not ubuntu Qt 5.5.1).  Only having an issue with map tiles not coming in (tile request error "problem with tile image") and QGC crashing when console view is selected.  I'll verify input video comes in tomorrow.

dogmaphobic, 2016-12-28:
> I successfully built QGC today on Kubuntu 16.04 with streamingvideo and uvc support

Did you make any modifications to `VideoStreaming.pri`? If so, would you mind sharing it?

> Only having an issue with map tiles not coming in

It sounds like google tiles is not working properly (they keep changing the API). Try using Bing and see if the problem goes away. This will determine if it is indeed a case of the google API going out of sync.

dogmaphobic, 2016-12-29:
@yzzir Any input on how you got GStreaming built on Ubuntu 16.04?

yzzir, 2017-01-03:
@dogmaphobic I use gstreamer for the ubuntu repos.  I verified that it is still working with master pulled today.  I didn't have to make any changes to VideoStreaming.pri.

DonLakeFlyer, 2017-01-03:
> It sounds like google tiles is not working properly (they keep changing the API). Try using Bing and see if the problem goes away. This will determine if it is indeed a case of the google API going out of sync.

@dogmaphobic Is there a way to create a unit test for this so we will know when they've changed things?

yzzir, 2017-01-03:
With regard to the tiles issue, I have always been trying to pull using Bing as they have a better hybrid view of stat+street view.  I believe e the issue is due to my work proxy.  I have it working on one PC (Ubuntu 14.04) but not on multiple others (Ubuntu 14.04 and Ubuntu 16.04) no clue why as I've imported all relevant CA certificates pertaining to my work proxy.  Also, the tile request issue doesn't appear when running qgroundcontrol from my home network (no proxy)

DonLakeFlyer, 2017-01-04:
@yzzir Can you enter a separate issue for your tile issue and put the detail there

DonLakeFlyer, 2017-01-04:
Ok, I'm not sure what do with 3.1 with respect to Ubuntu and gstreamer. Things like the AppImage don't seem to include gstreamer in it. Hence it is broken on 16.04 which is the new LTS so we should be able to run on that. Also the README instructions for installing gstreamer don't seem to work on 16.04 either. So pre-built archives and build yourself doesn't work either.

Linux is foreign to me so I'm going to need some help fixing this or my only option will be to turn off Video on Ubuntu. I really want to get 3.1 out the door since I'm heading on the road soon.

@dagar Can you help?

dagar, 2017-01-04:
Possibly, I've never actually used video in QGC. What's the quickest way to get a video source for testing? Is there a PX4 gazebo simulation or an easy way to stream from my own webcam?

DonLakeFlyer, 2017-01-04:
https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md shows a way to get a bouncing ball video stream

dogmaphobic, 2017-01-04:
> Is there a way to create a unit test for this so we will know when they've changed things?

In theory yes. Set the test to download a few, predefined tiles and see if they were downloaded successfully. It would be a matter of creating a function to load a tile from a specific provider given an x,y grid coordinate and zoom level. This is primarily an issue with google, which makes fetching tiles a game of whack-a-mole exactly for this reason.

DonLakeFlyer, 2017-01-04:
My current findings...

Installed the following packages:
libgstreamer1.0-dev
libgstreamer-plugins-base1.0-dev
libgstreamermm-1.0-dev

Changed in VideoStreaming.pri:
gstreamer-video-1.0 -> gstreamer-plugins-base-1.0

This then seems to at least get the correct headers but then it gets unresolved references in link. The references seem to be from things that should be in gstreamer-plugins-base-1.0. So not sure what to look for next.

DonLakeFlyer, 2017-01-04:
I'm think I'm going to go ahead with 3.1 release without this. It's no worse than it was before. For example AppImage on 14.04 doesn't really work either since it doesn't include the right gstreamer stuff either.

If we can figure it out we can patch.

ChristophTobler, 2017-05-12:
Any news on this? Was someone able to run QGC with GStreamer on Ubuntu 16.04? If so, what did you have to do?

uavcam, 2017-05-13:
Sure its working on 16.04. 

What you have to do: install the right gstreamer packages. See my original post

> On a ubuntu 16.04 64bit system the
    sudo apt-get install gstreamer1.0*

will fail because there is a i386 dependency somewhere. and even after installing all the packages by hand the libgstreamer-plugins* dev packages for the gstreamer-video-1.0 are missing

    PKGCONFIG += gstreamer-1.0 gstreamer-video-1.0

this needs libgstreamer-plugins* packages. do not know which one exactly - you guys will figure it out. will be one of the dev packages.

ChristophTobler, 2017-05-15:
@uavcam Thanks, but this is all quite vaguely... I was hoping for some command lines to install the packages

ChristophTobler, 2017-05-15:
```
list=$(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris)
sudo apt-get install $list
```
can be used in 16.04 since the only package with dependency issues is `gstreamer1.0-hybris`

uavcam, 2017-05-16:
apt-get install libgstreamer-plugins1.0*

could work. i use it without the 1.0 - but i think thats installing too much...

UavLabsColombia, 2017-05-30:
Hola, yo he solucionado este inconveniente en debian instalando el paquete 
sudo apt-get install libgstreamer-plugins-base1.0-dev

Saludos.

jaxxzer, 2017-06-21:
I'm not sure what this is all about, but the appimages downloaded from S3, as well as local builds have been working for me on ubutnu 14.04 and ubuntu 16.04 without any modification to the .pri or elsewhere in the repo. All I did was `apt-get install gstreamer1.0*`, per the current instructions. I suspect that the original post had a broken dependency, and that would be an issue outside of QGC or the current docs.

DonLakeFlyer, 2017-06-21:
Sounds good to me then

jaxxzer, 2017-07-14:
I did run into trouble with this today, after all. I'll see if I can come up with a definitive list of packages we need, but I did need to grab gstreamer1.0-plugins-bad (h264parse) and gstreamer1.0-libav (avdec_h264).

DonLakeFlyer, 2017-07-15:
Can you update README for video when you find it?

jaxxzer, 2017-07-20:
Yes

DonLakeFlyer, 2017-12-29:
Can any of the Linux folks compile this down to exactly how to install gstreamer onto 16.04? Need to get this straightened out for 3.3 release.

DonLakeFlyer, 2017-12-31:
The answer is in the thread above: 
```
list=$(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris)
sudo apt-get install $list
```

closed 2018-01-04 by DonLakeFlyer

FavorMylikes, 2021-03-02:
Is there any solution for centos 8?

BeardAndHarleyWojciech, 2022-09-12:
> https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md shows a way to get a bouncing ball video stream

The link is broken :(