mavlink/qgroundcontrol #4305: Seg fault running in VMWare with VideoReceiver::start().

gbmhunter, 2016-12-08:
QGroundControl (devel, 2016-12-06 nightly build) crashes on startup. The console prints:

`VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')`

The full console output is at the bottom of this post.

I am running QGroundControl inside VMware Fusion, with the following specs:

- Host = Mac OSX
- Client = Ubuntu 14.04
- 3D Graphics Acceleration = Yes
- Graphics Card = NVIDIA GeForce GT 750M

````
gbmhunter@ubuntu:~/Downloads$ ./QGroundControl.AppImage 
Settings location "/home/gbmhunter/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/gbmhunter/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed
Map Cache in: "/home/gbmhunter/.cache/QGCMapCache300" / "qgcMapCache.db"
Orphaned log file count 1
Orphaned log file "/tmp/FlightData914617.mavlink"
QGCMessageBox (unit testing) "Found unsaved Flight Data" "This can happen if QGroundControl crashes during Flight Data collection. If you want to save the unsaved Flight Data, select the file you want to save it to. If you do not want to keep the Flight Data, select 'Cancel' on the next dialog."
LinkManager mavlinkStatus 0xe87308 1 2
switch to mavlink 2.0 0xe87308 1 2
Temp log "/tmp/FlightData552460.mavlink"
ParameterManagerLog: Parameters loaded from cache /home/gbmhunter/.config/QGroundControl.org/ParamCache/1_1
Segmentation fault (core dumped)
````

dogmaphobic, 2016-12-08:
Based on the console output above, the seg fault occurs much later, after the vehicle is connected. I don't think it had anything to do with video. 

There was a crash on parameter handling for a bit yesterday but that has been fixed. I wonder if you got a version in between.

gbmhunter, 2016-12-08:
Ahh yes you are correct (I was thinking perhaps the fact the video didn't start caused a seg fault further down in the code). Using the latest nightly build fixes the issue.

Thanks!

closed 2016-12-08 by gbmhunter