mavlink/qgroundcontrol #4312: Multiple Video Feed Connections Crash

SteveJos, 2016-12-09:
Trying to show the video-feed window in another tab beside the flyview tab ends in crashing QGC with following error:

ASSERT: "m_materialType == MaterialTypeVideo" in file ../prodrone_qgc/src/VideoStreaming/gstqtvideosink/painters/videonode.cpp, line 55

The reason might be a second connection to the same video-feed, which is first created in the flyviewtab. If you disable the code that is showing the video-window in the flyviewtab, the custom video-feed window is working without the crash.

The code is:

`    FlightDisplayViewVideo {
                anchors.fill:   parent
                visible:        QGroundControl.videoManager.isGStreamer
            }
            //-- UVC Video (USB Camera or Video Device)
            Loader {
                id:             cameraLoader
                anchors.fill:   parent
                visible:        !QGroundControl.videoManager.isGStreamer
                source:         QGroundControl.videoManager.uvcEnabled ? "qrc:/qml/FlightDisplayViewUVC.qml" : "qrc:/qml/FlightDisplayViewDummy.qml"
            }`

aleixpol, 2017-03-09:
I ended up here because I'm having the same QtGStreamer crash. Good luck!

DonLakeFlyer, 2017-03-09:
We can really debug crashes in custom code. If you have a crash while running regular QGC then please report it in a new issue.

closed 2017-03-09 by DonLakeFlyer