mavlink/qgroundcontrol #4321: Mavlink inspector widget shows incorrect PWM values for servo_output_raw

kd0aij, 2016-12-15:
The analyze widget shows the correct values, but the inspector mostly shows 1500 for servos 1-4 (standard AERT plane). I have observed (rarely) the correct values displayed very briefly.

DonLakeFlyer, 2016-12-23:
Looks like a firmware problem:
* ArduPilot stack - good values
* PX4 stack - crap values (both inspector and analyze)

kd0aij, 2016-12-23:
Do both the inspector and analyze widgets display the same data? If so, what would explain the observed difference in behavior?

DonLakeFlyer, 2016-12-23:
They show the same data. If you are using an old build you may run into mavlink 2 problems with decoding which may be different between the two. But with latest daily it should be exactly the same. I can't repro the two things showing different values on daily build.

kd0aij, 2016-12-23:
Thanks, I'll assume the original issue was a mavlink 2 problem on an old QGC build and close this.

closed 2016-12-23 by kd0aij