mavlink/qgroundcontrol #4324: Crash on macOS after sleep/screensaver

szechyjs, 2016-12-16:
When coming out of sleep/screensaver the application freezes with spinning pinwheel and requires a force close.

Here is a portion of the error report:
```
Date/Time:       2016-12-16 16:19:39 -0500
OS Version:      Mac OS X 10.12.1 (Build 16B2657)
Architecture:    x86_64h
Report Version:  25

Data Source:     Stackshots

Command:         QGroundControl
Path:            /Users/USER/*/QGroundControl.app/Contents/MacOS/QGroundControl
Version:         3.0.2 ()
Parent:          launchd [1]
PID:             89978

Event:           hang
Duration:        1.30s (process was unresponsive for 82 seconds before sampling)
Steps:           13 (100ms sampling interval)

Hardware model:  MacBookAir6,2
Active cpus:     4

Fan speed:       1210 rpm

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  13  <truncated backtrace>
  13  semaphore_wait_trap + 10 (libsystem_kernel.dylib + 74838) [0x7fffa0c36456]
 *13  semaphore_wait_continue + 0 (kernel + 1144912) [0xffffff8000317850]

Process:         QGroundControl [89978]
Path:            /Users/USER/*/QGroundControl.app/Contents/MacOS/QGroundControl
Architecture:    x86_64
Parent:          launchd [1]
UID:             501
Task size:       98 MB
CPU Time:        0.220
Note:            Unresponsive for 82 seconds before sampling
Note:            3 idle work queue threads omitted
Dispatch Thread Soft Limit Reached:   64  (too many dispatch threads blocked in synchronous operations)

 Thread 0x7ac630           DispatchQueue 1           13 samples (1-13)         priority 46 (base 46)
  <thread QoS user interactive (requested user interactive), process unclamped, process received importance donation from WindowServer [188], process received importance donation from EmojiFunctionRowIM_Extension [1016], process received importance donation from com.apple.speech.speechsynthesisd [55549], IO tier 0>
  13  <truncated backtrace> 1-13
    13  semaphore_wait_trap + 10 (libsystem_kernel.dylib + 74838) [0x7fffa0c36456] 1-13
     *13  semaphore_wait_continue + 0 (kernel + 1144912) [0xffffff8000317850] 1-13

  Thread 0x7ac639           DispatchQueue 10          13 samples (1-13)         priority 31 (base 31)     cpu time 0.003
  <thread QoS default (requested default, override default), process unclamped, process received importance donation from WindowServer [188], process received importance donation from EmojiFunctionRowIM_Extension [1016], process received importance donation from com.apple.speech.speechsynthesisd [55549], IO tier 0>
  13  start_wqthread + 13 (libsystem_pthread.dylib + 12817) [0x7fffa0d26211] 1-13
    13  _pthread_wqthread + 1299 (libsystem_pthread.dylib + 14134) [0x7fffa0d26736] 1-13
      13  _dispatch_worker_thread3 + 99 (libdispatch.dylib + 15543) [0x7fffa0adacb7] 1-13
        13  _dispatch_root_queue_drain + 917 (libdispatch.dylib + 16537) [0x7fffa0adb099] 1-13
          13  _dispatch_client_callout + 8 (libdispatch.dylib + 8488) [0x7fffa0ad9128] 1-13
            13  _dispatch_call_block_and_release + 12 (libdispatch.dylib + 44895) [0x7fffa0ae1f5f] 1-13
              13  ___ZN19SpeechChannelHandle14HandleCallbackEPvb_block_invoke.134 + 426 (SpeechSynthesis + 29799) [0x7fff8a6e7467] 1-13
                13  DisposeSpeechChannel + 31 (SpeechSynthesis + 3969) [0x7fff8a6e0f81] 1-13
                  13  SpeechChannelHandle::~SpeechChannelHandle() + 90 (SpeechSynthesis + 18908) [0x7fff8a6e49dc] 1-13
                    13  usleep + 54 (libsystem_c.dylib + 526950) [0x7fffa0bc4a66] 1-13
                      13  __semwait_signal + 10 (libsystem_kernel.dylib + 106462) [0x7fffa0c3dfde] 1-13
                       *13  semaphore_wait_continue + 0 (kernel + 1144912) [0xffffff8000317850] 1-13

  Thread 0x7ac65f           DispatchQueue 10          13 samples (1-13)         priority 31 (base 31)     cpu time 0.003
  <thread QoS default (requested default, override default), process unclamped, process received importance donation from WindowServer [188], process received importance donation from EmojiFunctionRowIM_Extension [1016], process received importance donation from com.apple.speech.speechsynthesisd [55549], IO tier 0>
[... 45 lines cut ...]
  <thread QoS default (requested default), process unclamped, process received importance donation from WindowServer [188], process received importance donation from EmojiFunctionRowIM_Extension [1016], process received importance donation from com.apple.speech.speechsynthesisd [55549], IO tier 0>
  13  <truncated backtrace> 1-13
    13  __select + 10 (libsystem_kernel.dylib + 106318) [0x7fffa0c3df4e] 1-13
     *13  ??? (kernel + 5817008) [0xffffff800078c2b0] 1-13

  Thread 0x7ac680           Thread name "com.apple.CFSocket.private"            13 samples (1-13)         priority 46 (base 46)
  <thread QoS user interactive (requested user interactive), process unclamped, process received importance donation from WindowServer [188], process received importance donation from EmojiFunctionRowIM_Extension [1016], process received importance donation from com.apple.speech.speechsynthesisd [55549], IO tier 0>
  13  thread_start + 13 (libsystem_pthread.dylib + 12833) [0x7fffa0d26221] 1-13
    13  _pthread_start + 286 (libsystem_pthread.dylib + 14839) [0x7fffa0d269f7] 1-13
      13  _pthread_body + 180 (libsystem_pthread.dylib + 15019) [0x7fffa0d26aab] 1-13
        13  __select + 10 (libsystem_kernel.dylib + 106318) [0x7fffa0c3df4e] 1-13
         *13  ??? (kernel + 5817008) [0xffffff800078c2b0] 1-13

  Thread 0x7ac6aa           Thread name "com.apple.NSEventThread"               13 samples (1-13)         priority 46 (base 46)
  <thread QoS user interactive (requested user interactive), process unclamped, process received importance donation from WindowServer [188], process received importance donation from EmojiFunctionRowIM_Extension [1016], process received importance donation from com.apple.speech.speechsynthesisd [55549], IO tier 0>
  13  thread_start + 13 (libsystem_pthread.dylib + 12833) [0x7fffa0d26221] 1-13
    13  _pthread_start + 286 (libsystem_pthread.dylib + 14839) [0x7fffa0d269f7] 1-13
      13  _pthread_body + 180 (libsystem_pthread.dylib + 15019) [0x7fffa0d26aab] 1-13
        13  _NSEventThread + 205 (AppKit + 1656788) [0x7fff895877d4] 1-13
          13  CFRunLoopRunSpecific + 420 (CoreFoundation + 554260) [0x7fff8b8a7514] 1-13
            13  __CFRunLoopRun + 1361 (CoreFoundation + 556225) [0x7fff8b8a7cc1] 1-13
              13  __CFRunLoopServiceMachPort + 212 (CoreFoundation + 559156) [0x7fff8b8a8834] 1-13
                13  mach_msg_trap + 10 (libsystem_kernel.dylib + 74778) [0x7fffa0c3641a] 1-13
                 *13  ipc_mqueue_receive_continue + 0 (kernel + 894320) [0xffffff80002da570] 1-13
```

DonLakeFlyer, 2016-12-22:
Can you try the daily build?

DonLakeFlyer, 2016-12-23:
Also is this a 100% repro?

DonLakeFlyer, 2016-12-29:
@szechyjs Any help with testing?

szechyjs, 2016-12-29:
@DonLakeFlyer latest daily appears to be working.

DonLakeFlyer, 2016-12-30:
Thanks for reporting back

closed 2016-12-30 by DonLakeFlyer