mavlink/qgroundcontrol #4325: Android - Bluetooth will not connect through Dragon Link

wallbanger123, 2016-12-17:
Tower will connect to the flight controller through the Dragon Link v3 Bluetooth telemetry. QGroundControl will not connect.

What can I provide to help?

DonLakeFlyer, 2016-12-22:
Are you using a Daily build? I think Bluetooth was just fixed.

DonLakeFlyer, 2017-02-18:
@wallbanger123 Can you install the latest daily build?

wallbanger123, 2017-03-25:
It works on the PC/Mac versions, but Android (tried on multiple devices) it just crashes.

DonLakeFlyer, 2017-03-25:
Install the latest android daily build from here: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.apk. Bluetooth should be working with that build.

The Google Play listing for daily builds is not working correctly and is out of date.

closed 2017-08-07 by wallbanger123