mavlink/qgroundcontrol #4326: Parameter loading never completes over wifi on Android

kd0aij, 2016-12-19:
Progress bar stops at > 90% when connecting to Pixracer via wifi on Android 
Fails on both Nexus7 and MotoG3

Wifi connection works fine with QGC running on Ubuntu 15.04; parameters load quickly and completely.

DonLakeFlyer, 2016-12-22:
Daily or Stable?

kd0aij, 2016-12-22:
I think I'm in the beta testing group for Android

DonLakeFlyer, 2016-12-24:
I just testing this by installing the Daily Build from the app stores. Works fine on both PX4 and ArduPilot over WiFi. I fixed a number of lossy connection (like WiFi) parameter load bugs about 3 weeks ago. Can you verify that you are using latest daily build? Go to Settings/General. At the very bottom of the page you should see a line which shows you the build version. It should say Development and have a recent date. Last daily was from the 23rd. If that line isn't there you are either runnning a Stable or a really old Daily.

kd0aij, 2016-12-24:
I'm on the last daily build as of today. BTW, there's nothing lableled
"Settings" on the Android UI.

With a recent master installed on a pixracer, the progress bar stops just
short of the right edge of the screen, and the tablet is constantly saying
"communicati ~ communication regained".

The tablet reports that wifi signal strength is "excellent"

Just reverified that parameters load quickly and completely via wifi with
QGC running on my Linux desktop.

Until recently this tablet was my primary GCS for calibration and tuning of
small quads, so something must have changed in my Android setup.

DonLakeFlyer, 2016-12-24:
Settings is the Gear icon. 

If you are getting the commication lost/regained thing then your WiFi connection is flaky. If it's really flaky it's possible QGC can't complete param load. It should eventually time out completly and tell you that. Can you boot QGC with no Vehicle. Turn on ParameterManagerLog on Settings/Console, then connect your vehicle and when it stops for a while save the log an attach.

kd0aij, 2016-12-24:
Tried to save a log, but can't find it in the destination folder. There
appeared to be no way to name it, is the name generated magically?

kd0aij, 2016-12-24:
By "gears icon" do you mean the one that looks like a "Q"? That's the one with "general" and "console" behind it, at least on Android.

[image]

DonLakeFlyer, 2016-12-24:
Crap, sorry. Settings is the "Q" not the Gears which is Setup. Duh.

DonLakeFlyer, 2016-12-24:
> Tried to save a log, but can't find it in the destination folder. There
> appeared to be no way to name it,

Hmm, that dialog is kind of strange never used it before.  You need to add the filename at the top by adding a "/" and then the name. I'll put in and Issue for it.

kd0aij, 2016-12-24:
tried that, but it didn't appear to create a file anywhere

DonLakeFlyer, 2016-12-24:
Strange worked for me. May need to try again after I rework that dialog.

kd0aij, 2016-12-25:
Just tried the latest QGC on my MotoG3, and parameters loaded completely, but QGC still complains communication lost/regained continuously. WiFi analyzer app shows continuous max signal strength for pixracer though.

DonLakeFlyer, 2016-12-25:
If you are getting popups which says communication lost/regained that is coming from the Vehicle not QGC. QGC will only show "COMMUNICATION LOST" in the toolbar if it doesn't get a hearbeat for more then 3.5 seconds.

DonLakeFlyer, 2016-12-27:
Closing this out since parameter load is working ok on latest daily. Tested WiFi to Pixracer on OSX and Android and that was working fine as well. Likely something in your setup going wrong.

kd0aij, 2016-12-27:
Any ideas what might be wrong with my setup if it works on Linux, but not Android?

DonLakeFlyer, 2016-12-27:
This is on your MotoG2 right? That's going to have a crappier WiFi connection so it could be losing packets. Once it boots up and you've been to say Setup (Gear) and Fly (paper airplane) more than once, is it sluggush two move between those screens. It could be it's running too slowly on your phone to keep up but I've never seen that before.

Do you know how long it takes for the firmware to complain about communication lost? How much time in between heartbeats?

kd0aij, 2016-12-27:
The Nexus7 never had any performance issues, but fails to load parameters as of my last test. 
* Has something changed recently that would merit retesting?
* How would I measure the heartbeat interval on Android?

DonLakeFlyer, 2016-12-27:
Nexus7 stills fails on latest daily? If so need to re-open.

You'd need to measure heartbeat on the firmware side to measure GCS->Vehicle heartbeats. @LorenzMeier Do you know offhand how long before firmware complains about missing heartbeats?

DonLakeFlyer, 2016-12-27:
Assuming Nexus7 still screwed up

dogmaphobic, 2016-12-27:
Let me test this as well. I haven't booted the PixRacer in quite a while. What firmware should I flash to test?

DonLakeFlyer, 2016-12-27:
Mark is running master firmware

kd0aij, 2016-12-27:
Just retested Nexus7 with latest master and parameters loaded completely, but communication lost/regained every few seconds with signal strength -45dBm

dogmaphobic, 2016-12-27:
Ok, let me build master and flash it. I can test on several variations of "Nexus"...

DonLakeFlyer, 2016-12-27:
Mark: When you say communication lost/regained you are talking about message popus under the toolbar with the "X" in it right. Not the "COMMUNICATION LOST" messages which show up directly in the toolbar with the "Disconnect" button. Right?

dogmaphobic, 2016-12-27:
Crap, any idea what this is?
```
git submodule update
error: no such remote ref 44ad7e224c1ef17911ab8b4101fd624ad9ee4177
Fetched in submodule path 'NuttX', but it did not contain 44ad7e224c1ef17911ab8b4101fd624ad9ee4177. Direct fetching of that commit failed.
```

kd0aij, 2016-12-27:
try a make clean or distclean; NuttX changed recently

dagar, 2016-12-27:
@dogmaphobic if you don't have any changes in submodules try `make submodulesclean`.

DonLakeFlyer, 2016-12-27:
Yeah, it's stupid Git crap going berzerk. Any more questions? You seem surprised!

dogmaphobic, 2016-12-27:
Yup, that did it. Building and flashing...

kd0aij, 2016-12-27:
as of today, the MotoG3 doesn't connect at all
must have pushed the wrong button

kd0aij, 2016-12-27:
BTW, I used QGC to flash master; assuming it grabs a recent CI build

dogmaphobic, 2016-12-27:
I can't even get the tablet to connect to the PixRacer's WiFi. Android 7. It keeps saying there is no Internet connectivity and refuses to connect. Plugging it using USB (so I can change it to use *station* mode instead) but it is not showing up in QGC (no *WiFi Bridge* component is listed under Vehicle Setup).

DonLakeFlyer, 2016-12-27:
Ah, that's an android 7 only problem that I though you could get past somehow (refusal to connect to WiFi with no internet). We need to duplicate how Tower gets past this. There is an Issue on this somewhere with details. Looks like it needs to be fixed for 3.1.

DonLakeFlyer, 2016-12-27:
@kd0aij Can you try saving a console log with ParameterManagerLog turned on with your Nexus 7 again? You'll get a strange file dialog. In the left panel you should be able to click on either Documents or Desktop. And then add a "/" followed by a file name and it should save there.

kd0aij, 2016-12-28:
Save to Documents worked:
https://drive.google.com/open?id=0Bw3digSMQXDuZGI2N19xcDRwM2c

dogmaphobic, 2016-12-28:
Booted Nexus 9 and was able to duplicate the issue (did not load all parameters and it keeps yelling at me telling *communication lost* followed by *communication regained*). Deleted some of my comments above as they only added noise. Let me run some more tests.

dogmaphobic, 2016-12-28:
More than half the packets from the WiFi Bridge are being lost as you can see in the screen grab below. I have the same behavior when connecting with the Desktop. I wonder if the data rate is set too high. I will see if tweaking it makes it any better.

[image]

DonLakeFlyer, 2016-12-29:
Are those numbers from sequence number gaps? I see the same thing on my Samsung Galaxy running 5.02. Although I don't see the communication lost thing.

dogmaphobic, 2016-12-29:
Yes, something is royally messed up. This was working fine at some point. The lost count is real tough it doesn't match the rest because I had reset the counters. It seems it all starts fine, with little to no drops for a while. Then you can see "pauses" and the drop count starts to pile up. I wasted a whole lot of time chasing the Android 7 "No Internet" issue (with no good solution) today. For this WiFi UDP problem, I need to setup a full debug system, with a different ESP8266 unit with its own debug output running in tandem. Only then I will be able to track down where the bottleneck is happening. This is assuming that @kd0aij is running the ESP-01, right? That's what I was testing with.

dogmaphobic, 2016-12-29:
Also note that the WiFi Bridge does not support MAVLink V2. I wonder if that has anything to do with this.

DonLakeFlyer, 2016-12-29:
MAV_PROTO_VER param controls mavlink 2 on/off. QGC won't send MAV 2 unless it see it coming from the Vehicle.

kd0aij, 2016-12-29:
@dogmaphobic My ESP8266 modules are the same old ones that I first tested on pixracers. And they were both working fine till recently.

dogmaphobic, 2016-12-29:
> My ESP8266 modules are the same old ones that I first tested on pixracers. And they were both working fine till recently.

Right, that's what I thought. I've been testing with the same thing and yes, it was working fine the last time I used it. I will do some more thorough testing tomorrow.

dogmaphobic, 2016-12-29:
@DonLakeFlyer changes last night seem to have fixed it all. I now see close to zero lost messages. @kd0aij Can you try QGC's current master as well?

kd0aij, 2016-12-30:
Will these changes be in the Android beta yet? I'm not set up to build for Android myself.

DonLakeFlyer, 2016-12-30:
Yup. Should be there. Just double-check daily build version in Settings/General to make sure you have new build.

kd0aij, 2016-12-30:
Seems to be OK now; no "communication lost/regained" messages so far, but
losing about 20 messages/second according to the WiFi Bridge Status display.

DonLakeFlyer, 2016-12-30:
Ok, thanks for testing Mark. This specific issue is fixed now.

@dogmaphobic Your call on whether you think 20/second is acceptable.

closed 2016-12-30 by DonLakeFlyer

dogmaphobic, 2016-12-30:
> Your call on whether you think 20/second is acceptable.

20/second doesn't mean much. I would say up to 10% loss is acceptable. I'm seeing around 1% overall.

kd0aij, 2016-12-30:
What are the variables involved, if an absolute loss rate "doesn't mean much"?

dogmaphobic, 2016-12-30:
20/second depends on how many messages were sent within that one second. If you received 20 messages and lost 20, it's 100% loss. If you received 2,000 messages, it's 1% loss. From empirical experience, 10% is acceptable (20/200)

kd0aij, 2016-12-30:
duh

[image]

dogmaphobic, 2016-12-30:
A bit high but it should be fine for telemetry. Are you using a 2.4GHz RC radio? I'm not seeing anywhere near that much loss (no RC connected). The PixRacer/ESP8266 link (Serial) loss is quite high too. Try reducing the data rate (not the baud rate). Something like:

`mavlink start -r 10000 -m config -b 921600 -d /dev/ttyS0`

And see if it makes a difference.

LorenzMeier, 2016-12-30:
@kd0aij The ESP8266 has no flow control handling. So you need to (as @dogmaphobic says) adjust all data rates carefully so no buffer overflows.