mavlink/qgroundcontrol #4328: Crash on android when direct usb connection to board

icesword001, 2016-12-20:
On android, In the beginning，connection QGC is successful, everything is all right; 
However, after waiting for tens of seconds to minutes, the QGC would Crash; 
I found, debugging is a collapse in the class, but the reasons is unknown

DonLakeFlyer, 2016-12-22:
Are you using an Android daily build? There was a crash after some amount of time was fixed in Daily.

icesword001, 2016-12-23:
I use the QGroundControl v3.0.2  package（from github）; 
There are the same problems.

DonLakeFlyer, 2016-12-23:
3.0.2 is a stable buid not a daily build. To get daily build you need to sign up for the android beta: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/download_and_install.html

DonLakeFlyer, 2016-12-25:
@icesword001 I'll re-open if it is still crashing on a Daily Build.

DonLakeFlyer, 2017-05-23:
Retested on daily build. Working fine

JonReacher, 2017-07-27:
@DonLakeFlyer I'm still having this issue on a stable build for months, even to the latest 3.2.2. Haven't ever been able to connect for more than a 10 seconds directly to the board.

LorenzMeier, 2017-07-27:
@dogmaphobic maybe related to what you found?

mhkabir, 2017-07-27:
Can confirm that this is still an issue.

dogmaphobic, 2017-07-27:
I found a nasty bug when QGC receives a message with a message ID > 255. It causes buffer overrun and all things go haywire after that. The behavior is random as the affected places change based on the message and platform. I will push the fix shortly.

DonLakeFlyer, 2017-08-12:
I can reproduce this. But I'm having probelsm with my tablet where I can't get it to go into wireless debugging mode to be able to catch the crash.

birchera, 2017-08-17:
We are also experiencing this bug. I tracked it down to happen when clearing the QGCRingBuffer in the serial read functionality. Specifically it seems that [clearing](https://github.com/mavlink/qgroundcontrol/blob/master/libs/qtandroidserialport/src/qserialport_p.h#L194) the remaining first QByteArray in the buffers list produces the error. However, no attempt to guard it with the obvious checks solved the issue.

I attached the stack trace below.

[SerialStackTrace.txt](https://github.com/mavlink/qgroundcontrol/files/1231826/SerialStackTrace.txt)

DonLakeFlyer, 2017-08-17:
@birchera I can repro this, but I haven't been able to get it into the debugger. Are you using QtCreator to debug?

birchera, 2017-08-17:
Normally I do, but without the cable available, I also could not yet attach the debugger. So I only have the adb shell at hand.

PR #5562 (2017-08-12): Stable: Buffer overrun in UAS.cc
Cherry picked #5504 from master to try to fix #4328. Unfortunately it does not fix #4328. But is is a real problem nonetheless.
PR #5638 (2017-09-08): Android: Serial port crash fix
Related to #5637. Fixes #4328
closed 2017-09-09 by DonLakeFlyer