mavlink/qgroundcontrol #4329: Weird behavior in Comm Links (Not user friendly)

samacumen, 2016-12-20:
Hi,

The issue I want to report is about creating a new connection via 'Comm Links' under Application Settings (1st Tab) on QGC.

I see that there is always a 'default UDP link' running and it figures out what the remote port should be for sending packets (which is kind of smart, but annoying from a user point of view). However, once I disconnect a new entry is automatically added. What I do not understand is why can't the user 'delete' the previous entries at will? Restart the QGC and all the previous entries disappear.

See the two images below, on creating new (automatic) connections and restarting QGC.

[image]
[image]

2) Also, If a user creates an entry for connection, that should be taken up by default the next time. However, the QGC has its own logic behind to set up a UDP connection that it likes - kind of annoying!

Basically, this part of the interface is important and it should be made more user friendly.

Regards,
Shyam

DonLakeFlyer, 2016-12-23:
This link is no longer shown in the list since it is an dynamic AutoConnect link. If you want to create your own UDP link with different settings than the standard UDP AutoConnect link you need to first turn off UDP AutoConnect from Settings/General page.

closed 2016-12-23 by DonLakeFlyer

samacumen, 2016-12-23:
Thanks @DonLakeFlyer . That answers part of my questions. Its good that we can disable the autoconnect.

However, when the autoconnect is made, there must be a dedicated link that user should not be able to disconnect, am I right? What is the point of disconnecting one and a new one (button) appears automagically below it - which is what is kind of annoying. Most users may still prefer the autoconnect. I know there was a long discussion on the forums regarding this sometime ago with Lorenz Meier and your team. I saw that some commits have been made to the QGC, does that fix any of these concerns?

Thanks for looking into the issue reported.

DonLakeFlyer, 2016-12-23:
> What is the point of disconnecting one and a new one (button) appears automagically below it

That will no longer happen. The AutoConnect UDP link no longer shows up in the list so you can't disconnect it. The only way to disconnect the UDP AutoConnect link is by turning off AutoConnect for UDP in Settings/General.

samacumen, 2016-12-23:
Ok, you mean to refer to "Default UDP Link" button no longer shows up in the list, am I right? Not the 'AutoConnect UDP link'.

Also, the issue I referred to was [https://github.com/mavlink/qgroundcontrol/issues/1681](url) kind of related to this. Is that also resolved?

DonLakeFlyer, 2016-12-23:
> you mean to refer to "Default UDP Link" button no longer shows up in the list

Correct.

Not what you mean with respect to #1681. That is the discussion around the AutoConnect implementation which was completed ages ago.

DonLakeFlyer, 2016-12-23:
@samacumen Do you have a 100% repro set of steps for this? I'll try some things, but easier if you already have it.