mavlink/qgroundcontrol #4333: Parameter Profiles (Feature Request)

tops4u, 2016-12-22:
In my Opinion it would be nice to be able to have different Profiles which could be stored on the Flightcontroller for various Tasks (or in Ground Station).

Currently I have not found such a function in PX4 nor QGC (Tried to search GIT too - apologies if this was already asked).

For example for a specific type of mission I would like to use a Ground LIDAR together with a Cruise speed of max 20kmh, smooth PID Values and a fixed YAW Heading in Mission Mode and Failsafe Action to RTL

In another type of mission I would like to set max Speed to 50kmh disregard the LIDAR, have YAW turn in direction to the next Waypoint and Failsafe Action to Land with a Geofence of 1000m and moderate PID Values due to different Payload.

Currently all the values have to be changed by Hand. Instead a Profile Switching (by MavLink Command or external Switch) would be desirable as it has i.e. been implemented in the BaseCam Gimbals where different Profiles can be stored dependent of the attached Camera.

Originating from: https://github.com/PX4/Firmware/issues/6138#issuecomment-268817764

DonLakeFlyer, 2016-12-23:
This ability currently exists in mission files. It is currently only used for GeoFence settings. Also there is no UI to add them so you need to edit the file manually. The parameters are changed when you load the mission. In 3.2 we can look at making it more user friendly.

I need to double-check that round tripping these params works if you are using them outside of geofence settings. Also need to document the format.

tops4u, 2016-12-23:
Are the Parameters only active during Mission Mode or once the Mission is loaded they overwrite previous Parameters? 
I'm still in favour of an FC Internal Solution for this by using kind of Profiles.

DonLakeFlyer, 2016-12-23:
Once the Mission is loaded they overwrite previous Parameters. Anything else would require firmware changes. If you want them only when mission is running then you are back to needing firmware changes.

DonLakeFlyer, 2016-12-27:
Also I was wrong in that parameters are only support by GeoFence and Rally loading, not by Mission fiules.

tops4u, 2016-12-27:
I'll reopen this Issue within PX4  then.

closed 2016-12-27 by tops4u