mavlink/qgroundcontrol #4335: 3DR radio firmware: unable to put into command mode

AndKe, 2016-12-22:
Plug in your device via USB to start firmware upgrade.
Found device: SiK Radio
Putting radio into command mode
Error: Unable to put radio into command mode
Upgrade cancelled
------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.

AndKe, 2016-12-22:
unable to reproduce reliably anymore, closing for now

closed 2016-12-22 by AndKe

DonLakeFlyer, 2016-12-23:
Radio upgrade can be a bit flaky. Something like 5-10% failure rate on getting the radio into the right mode. I was never able to figure out a 100% reliable way to make it work.

AndKe, 2016-12-23:
in your experience, is there a difference between v1 and v2 radios (there
should not be)  , I did not reflash that many, simply because I am using
en/decrypting firmware, and need to flash back later.
-I might compare a few applications/methods using a logic analyzer, that
should reveal any difference.

DonLakeFlyer, 2016-12-23:
Not sure. Never looked at it in that level of detail.

blandry, 2018-03-06:
I know this is old but if someone stumbles on this, try turning off the other radio (ie the one probably on your drone).

AndKe, 2018-03-06:
this has been fine for a long time now, the real remaining issue now is:
https://github.com/mavlink/qgroundcontrol/issues/4336

blandry, 2018-03-06:
I believe you. But it just happened to me (perhaps on an old radio) so I'm documenting what worked for me

DonLakeFlyer, 2018-03-13:
> I know this is old but if someone stumbles on this, try turning off the other radio (ie the one probably on your drone).

Interesting. I wonder if this really fixed the problem. I've found that flashing will fail to put into command mode intermittently at a low failure rate. I've never been able to make it work 100%. Did you have 100% failure until you did this?

PaulStubbs, 2019-10-28:
I get this error in QGroundControl (Unable to put radio into command mode) when one of the 2 radios is already plugged in. Make sure both radios are unplugged before upgrading the firmware.

myself248, 2020-06-28:
Just a data point for anyone else reading this: Flashing the drone-side radio may succeed or fail depending on which USB-UART adapter you're using! 

I had nothing but frustration with both CP2102 and CY7C6521x adapters, despite confirming with an oscilloscope that everything seemed to be talking appropriately. Got the "unable..command mode" error every time. Switched to a name-brand FTDI, and it just worked.

I'm not sure if QGC is doing something weird with the UART, but if there's interest I could try to narrow it down. (QGC 4.0.6 on Ubuntu 18.04.4)

danylevskyi, 2021-12-01:
I just switched TX and RX pins and it worked for FT232RL.
RX -> RX
TX -> TX

MiangChen, 2024-11-14:
> I just switched TX and RX pins and it worked for FT232RL. RX -> RX TX -> TX

This is helpful， thanks