mavlink/qgroundcontrol #4336: 3DR radio flashing: please let user choose firmware, ask for confirmation.

AndKe, 2016-12-22:
For consistent experience, when 3DR radio is about to be flashed: please make the behavior more like Ardupilot flashing: 

Ask user for confirmation, and offer to flash custom file, instead of just flashing right away.

DonLakeFlyer, 2016-12-23:
Also when choosing Custom Firmware for PX4/ArduPilot radio buttons should disable or hide so user isn't confused.

AndKe, 2018-02-24:
this is still an issue, connecting a radio, means flashing it with current SiK , instead of at least confirming that the user want that, or offer to choose a firmware file..

myself248, 2020-07-03:
> Please unplug your Pixhawk and/or Radio from USB.
> Plug in your device via USB to start firmware upgrade.
> Found device: SiK Radio
> Putting radio into command mode
> Rebooting radio to bootloader
> Connected to bootloader:
> Version: 0
> Board ID: 130
> Flash size: 0
> Unable to find specified firmware for board type
> Upgrade cancelled
> -------------------------------------------

Can the check for firmware be done in advance, and refrain from even putting the radio into bootloader mode if there isn't an appropriate firmware for it? Or, if firmware isn't found, can the radio be taken back out of bootloader mode and restored to functionality? What it's doing right now is just bricking radios and I need to go dig out a Windows box with MP to unbrick them.

(The radio in the above example is an RFD900+ by the way.)

DonLakeFlyer, 2020-07-03:
> What it's doing right now is just bricking radios and I need to go dig out a Windows box with MP to unbrick them.

You should just need to remove power from them and they go out of command mode.

myself248, 2020-07-03:
> You should just need to remove power from them and they go out of command mode. 

I wish, but the red LED is on solid no matter how many times I power cycle it, until I go find something with valid firmware for it. Numerous similar tales of woe over here: https://discuss.ardupilot.org/t/3dr-radio-in-bootloader-mode-after-failed-firmware-update/14911

DonLakeFlyer, 2020-07-03:
> Numerous similar tales of woe over here: https://discuss.ardupilot.org/t/3dr-radio-in-bootloader-mode-after-failed-firmware-update/14911

So three years ago for the forum post and nobody bothers to notify QGC dev of the problem. Sigh!

DonLakeFlyer, 2020-07-03:
> Can the check for firmware be done in advance, and refrain from even putting the radio into bootloader mode if there isn't an appropriate firmware for it?

No. The id needed to know which firmware to flash comes from the bootloader.

> Or, if firmware isn't found, can the radio be taken back out of bootloader mode and restored to functionality?

Yes. Working on that.

DonLakeFlyer, 2020-11-08:
I believe this is supported now but SiK Radio flashing in general seems busted right now. Will verify once that's fixed.

destroyo123, 2024-09-13:
I am still having this issue. Qgroundcontrol updated my SiK radio firmware to a version incompatible with my air module so now they won't pair.

But Qgroundcontrol won't let me choose the older/compatible version to reupload to the ground link, so I can't use my telemetry radio.

closed 2026-10-05 by github-actions[bot]