mavlink/qgroundcontrol #4343: QGC crashes when sending Vehicle MAVLink message on a link! (at unexpected times)

samacumen, 2016-12-23:
I would like to bring to your notice that QGC crashes at unexpected times when using Serial/or multiple links. I recently figured out why and where this happens and the root cause. The culprit is the way we have written the function priorityLink() and its usage while sending mavlink messages.

If AutoConnect UDP is used, there will be at least one link in the **Vehicle::_links** Qlist. When the serial USB is connected, there is this serial link. What this logic of priorityLink() does is it gives Serial the priority and if it is not found, the first link in the _links list is returned.

```
/// @return Direct usb connection link to board if one, NULL if none
LinkInterface* Vehicle::priorityLink(void)
{
    foreach (LinkInterface* link, _links) {
        if (link->isConnected()) {
            SerialLink* pSerialLink = qobject_cast<SerialLink*>(link);
	    if (pSerialLink) {
                return link;
            }
        }
    }

    return _links.count() ? _links[0] : NULL;
}
```

**Two problems:**

1) return _links.count() ? _links[0] : NULL; 
When no vehicle is connected, we disable the QGC GUI - so we are safe. But during certain heavy processing, like sending a big mission to the vehicle, if the link (serial/other) is disconnected - the above could/is certainly NULL. Consequences? Crash of QGC, as we use prioritylink()->mavlinkChannel() while sending each message. Access on NULL instance!

2) If there are more than 1 links (apart from serial, say a UDP and a TCP i.e. Serial, UDP, TCP in the list) for the same vehicle (which is allowed) - which link must be chosen if Serial is not available? The first in the list after Serial, i.e. UDP -> does not make much sense. This indicates the TCP link will never be used.

Somehow it feels to me that the implementation of PriorityLink() is incomplete. Suggest to at least place a NULL check apart from link not being connected while sending ANY Mavlink message to the vehicle. 

PS: Also, notice that isConnected() for UDP link will never be 'false' (unless user disconnects manually) and isConnected() for Serial will usually not occur (as the link object is lost immediately on a disconnect).

@LorenzMeier , @DonLakeFlyer

LorenzMeier, 2016-12-23:
@dogmaphobic Are you able to come up with a patch? This might fix a plethora of issues / crashes.

dogmaphobic, 2016-12-23:
> Are you able to come up with a patch? This might fix a plethora of issues / crashes.

Well, it's a bit more than a *patch*. There are 81 references to `priorityLink()` and none of those calls ever check to see if the link is NULL. That is something that needs addressing. No call that receives a pointer should ever blindly use it, without checking for NULL first, no matter how remote the chances for a NULL are. With that said, it IS remote as it can only happen on corner cases. Yes, it should be fixed but how often are you connected to a vehicle with two or more links?

I never liked that whole dual link thing to begin with but I understand the need to handle it. The entire *link* system became a patch work over the years. I will *patch* it but we should start thinking about *rethinking* the whole link model.

The way I see it, when you plug something using a USB cable, you are doing firmware maintenance. Nothing else should be connected. That's the only time you end up with dual links, which I think should not be allowed at all.

LorenzMeier, 2016-12-23:
Dual links are pretty standard - e.g. a radio module and Wifi. People use them to retain command & control if the main link fails. That said QGC is lacking a suitable QoS mechanism. The firmware side on the other hand has it.

Why this works in practice is that QGC does not send a whole lot of data and the Firmware side QoS takes care of scalability. Which makes dual links extremely useful for development and industrial applications, which is the only fields where QGC really is reasonably strong as a tool right now.

Or in other words - relevant and used.

samacumen, 2016-12-23:
@dogmaphobic I agree with your statements. 

About the corner cases, well it could happen (and it has multiple times) under three different scenarios:
1) Plugging on/off USB while some (mavlink message passing) is still on - e.g. uploading a mission or during packet loss (e.g. long USB cable)
2) Pixhawk is rebooted/sudden power turn off - leading to QGC crash (in certain cases).
2) Using a remote link (tcp), and it has been disconnected - leaving no links in the list - and the user is somehow able to send a message on that link.

I looked at the code again, I see three possible solutions to this:
1) The least we must do is do a NULL check on all places with return value of priorityLink(). 
2) Throw an exception from within priorityLink(), if no links exist in the list. We disable certain parts of GUI from user interaction and cleanup (as there is no vehicle anymore). Ideally, this exception should never be thrown from a design perspective.
3) Emit a signal and handle it (though option 2) seems nicer).
 
I like the dual link logic. I am using the Vehicle on a UDP link - so that makes it one link. But it is quite possible that myself, as a user want to have a 2nd link for some purpose (if the first one fails).

Further, the way IsConnected() is handled in different classes (UDP, TCP, Bluetooth, etc. inheriting from LinkInterface) is a bit questionable. Probably, will raise another issue instead of deviating from this one.

For me, I have been working on QGroundcontrol for a quite sometime now and understand how the Vehicle and Link has been is handled - and I think the overall design of link manger (from a software perspective is pretty good) - expect for a few things are needs addressal and probably a redesign in some cases.

DonLakeFlyer, 2016-12-23:
Thanks @samacumen. I'll take a look through this. Multiple links gets nasty very quickly. Thought we have most of it wrung out but guess not.

The Vehicle::priorityLink implementation is meant to give priority to non WiFi links. It was originally coded to force to direct USB connections but that was causing problems. Can't remember why now. Would have to look a the pull request history.

Basically this is a shutdown problem with links. For things like missions (and parameter loading and geofence loading) there is a state machine with timers that go off. Those state machines need to be cleared/stopped when the link goes away. Otherwise you can end up with ordering problems and attempts to send when no links are available. Generally for normal mavlink message traffic this isn't a problem because as you say the UI goes away when a Vehicle goes away so you can't end up with a mavlink message from a UI action trying to go out when the link goes away.

Multiple links are only used for incoming messages. There is no such things as sending on multiple links when talking from QGC->Vehicle.

Also your statement: "If AutoConnect UDP is used, there will be at least one link in the Vehicle::_links Qlist.". Is not quite correct or maybe depends on what you meant. A link is only associated with a Vehicle when a Vehicle shows up as communicating through that link. So if your vehicle is not communicating via UDP even though there is an AutoConnect UDP link always hanging around it will not be associated with the Vehicle.

@dogmaphobic Let me look at this. Multiple links is a big can of worms you don't want to get in the middle of.

DonLakeFlyer, 2016-12-23:
The reason why this happens is because of the multi-phase vehicle delete mechanics to get correct Qml object disconnects. Basically when the last link goes away, the Vehicle is not destoryed immediately. This in turn allows additional timers to go through which can end up sending messages.

DonLakeFlyer, 2016-12-23:
> That said QGC is lacking a suitable QoS mechanism.

What would be a suitable QoS mechanism? QGC currently switches links for outgoing traffic if it stops getting heartbeats through a link in a multi-link use case.

DonLakeFlyer, 2016-12-23:
@rjehangir Possibly your problem as well.

samacumen, 2016-12-23:
@DonLakeFlyer Thanks for your comments. My statement about the AutoConnect UDP being used leading to atleast one link in the Vehicle::_links Qlist in indeed not completely correct. It gets filled in only when Vehicle connection is alive - I agree!

I don't have the exact 100% reproducible steps at the moment. I use QtCreator and often debug the code, and have seen the exception at prioritylink()->mavChannel() (when the return from prioritylink() becomes NULL). I must say, that it could be a bit hard to reproduce that's why I wrote at unexpected times - but I can assure you that I have seen this happening multiple times at the same location - so I can guarantee that the link in certain situation does become NULL ;)

DonLakeFlyer, 2016-12-23:
Do you remember whether the call stack of the crash led you to one of the *Managers: ParameterManager, MissionManager, GeoFenceManager, RallyPointManager. If my theory is correct, that is where it would end up.

DonLakeFlyer, 2016-12-23:
My plan to fix this is:
1) Add a shutdown() method to the *Managers which gets called when all the links go away. This will then close out any timers they may have which could in turn caused additional messages to get out the pipe at the wrong time.
2) Need to shut down command retry timer as well since this could fire at bad time.
3) Change vehicle->priorityLink()->mavlinkChannel() usage to something like vehicle->priorityLinkMavlinkChannel() which handles NULL link check and returns bogus mavlink channel in that case.
4) Vehicle::_sendMessageOnLink already does null link check so that cauterizes any crashes from getting through in there.

This way you don't need if (priorityLink() == NULL) all over the place and it should still prevent crashes due to shutdown ordering.

samacumen, 2016-12-23:
Also, one more thing I noticed regarding disconnections and heartbeats:

1) **SERIAL USB**: When USB is disconnected/plugged off - we will never end up firing the Vehicle::connectionLostTimeout( ) (OK, can practically happen, but it 99.9% of cases it should not with a good cable/min. packet loss, etc.). The Vehicle link is lost immediately after USB is disconnected on Serial. The vehicle object - thus the _links list (also part of vehicle) is cleared.

2) **UDP Connection:** In case of a UDP connection, if we once establish a remote connection to a vehicle on this link - the vehicle object is created. However, this vehicle object would not get destroyed (when say, the vehicle goes away or the communication at the vehicle side dies off for some reason). We would always end up in this recurring Vehicle::connectionLostTimeout( ) timer event. Therefore, from the perspective of QGC, the vehicle is still very much alive - which may not actually be the case. So, the vehicle object is retained and also the link is/may still be part of _links list.

Is the handling of the above situation intended/as designed?

DonLakeFlyer, 2016-12-23:
> When USB is disconnected/plugged off - we will never end up firing the Vehicle::connectionLostTimeout( ) (OK, can practically happen, but it 99.9% of cases it should not with a good cable/min. packet loss, etc.). The Vehicle link is lost immediately after USB is disconnected on Serial.

That is not what happens on Windows OS. On Windows you don't lose the connection when you pull the plug on usb cable. The serial port is still there and connected, it's just in a zombie state. Hence you need to connectionLostTimeout as well to make things work across all OS.

For #2 you can't tell the difference between a temporary loss of communication with a vehicle and/or the vehicle was just turned off/disconnected in all cases across of OS. Hence the connectionLossTimeout which then notifies the user and lets them disconnect manually if they want. If the users decides to manually disconnect that is when the Vehicle goes away.

All of the above is correct behavior.

samacumen, 2016-12-23:
@DonLakeFlyer Solution looks good to me as long as all connections are gracefully closed. Its nice that we can place the checks at one point instead of those 81 - as with more features/custom mavlink additions usage will only increase! Well, now I remember the issue also being notice by one of my colleagues when he was at the test field and the QGC crashed - luckily he was still connected on the USB ;) But hope your solution will do the trick :+1:

> Do you remember whether the call stack of the crash led you to one of the *Managers: ParameterManager, MissionManager, GeoFenceManager, RallyPointManager. If my theory is correct, that is where it would end up. 

It was certainly not GeoFenceManager, RallyPointManager, ParameterManager. But could be MissionManager - but not exactly sure. Could trace more closely when I see it next time. I didn't have a breakpoint at that place but only that the exception was seen there - the stack trace was showing up the assembly trace or moc rather than code functions AFAIK.

Thanks for the reasoning behind the connectionLostTimeout( ). I use Linux, departed long away from Windows - so not much idea about the latter.

DonLakeFlyer, 2016-12-24:
FYI: This bug was introduced with the MAVLink 2.0 changes. Those changes required pulling out the priorityLink()->mavlinkChannel() calls from Vehicle::sendMessage where it was cauterized by the NULL check and peppering the usage throughout the codebase. I'm going to try to figure out some way to put it back the way it was before since that was working fine. This crash will only be in daily builds.

samacumen, 2016-12-24:
Thanks @DonLakeFlyer , good to note that you have figured out the bug! I have worked with both Mavlink versions, so 'just curious' as to why the Null checks were removed (I mean the how does the different versions affect the logic here)? :) And what do you mean by > this crash will only be in daily builds?

DonLakeFlyer, 2016-12-25:
No null checks where removed. The position where you need to call LinkInterface::mavlinkChannel changed. Used to be in a single place behind a null check in Vehicle:sendMessage. Now it needs to be in all the *_pack_chan mavlink channels which are all of the codebase.

The mavlink 2.0 changes are new and have not yet been released to stable version.

closed 2016-12-27 by DonLakeFlyer