mavlink/qgroundcontrol #4344: Clicking on QGC icon twice displays an empty white screen

samacumen, 2016-12-24:
I thought to report this issue as well. Clicking on the top-left QGC (Q) Application settings icon twice makes the screen go white/blank and non-responsive. The user needs to click on some other icon (e.g. Navigation map icon) and then go back to the (Q) Application settings.  Don't know what this means, but guess it is an issue, even though minor it may be.

Click on the (Q) Application settings icon:
[image]

Click again on the (Q) Application settings icon:
[image]

I haven't tested this on a tablet which may be more sensitive to this happening, but this occurs on Linux (Ubuntu). But guess its' a general issue.

DonLakeFlyer, 2016-12-24:
Should already fixed in latest daily builds.

DonLakeFlyer, 2016-12-24:
Reopen if not

closed 2016-12-24 by DonLakeFlyer

samacumen, 2016-12-24:
Sure, thanks. I will check it with the latest build and get back if not.