mavlink/qgroundcontrol #4355: Android 7 refuses to connect to WiFi with no internet on it

DonLakeFlyer, 2016-12-27:
Some possible Tower details here: https://github.com/mavlink/qgroundcontrol/issues/4222

dogmaphobic, 2016-12-28:
Temporary solution for those with adb installed:

Run
 `adb shell settings put global captive_portal_detection_enabled 0`

* "Forget" the WiFi Bridge IP
* Reboot everything
* Connect again to the WiFi Bridge (SSID: PixRacer, PWD: pixracer)

It should now connect and work. Looking for a programmatically way to do this.

DonLakeFlyer, 2017-01-03:
@dogmaphobic Are you still working on this one? Wondering if I needed to start searching the Tower code. Have you verified that Tower does the right thing here?

dogmaphobic, 2017-01-03:
The setting required to make Android use WiFi when it cannot find "google.com" is at the system level. At this point, my Nexus tablet running Android 7 is already "fixed" (manually). I would have to revert the "fix" and see how Tower behaves, though I don't have anything to connect to (no Solo or anything APM for that matter).

The same would be true for apps such as the GoPro app, which connects to the camera's own AP, but in that case, we don't have access to the code.

The larger question: Is anything other than an updated Nexus running Android 7? Has Samsung updated to Android 7 already?

dogmaphobic, 2017-01-03:
Note that this issue is different from the one described in #4222. That one deals with the case where you have both Cellular and WiFi. If the WiFi link doesn't have Internet, Android will route everything through Cellular. Based on the documentation I found, it's a "feature" of Android 5.x and greater. That's the work around I've attempted with #4380.

This issue (Android 7) is that Android won't ever connect to a WiFi AP if it can't find google.com regardless of the state of your cellular link (I don't even have it on my Tablet).

DonLakeFlyer, 2017-01-03:
Not going to wait on 3.1 for this. If it is even fixable! It's also not a new thing so we'll move forward without it.

closed 2026-10-05 by github-actions[bot]