mavlink/qgroundcontrol #4361: Please allow battery to be monitored by voltage, not percent.

AndKe, 2016-12-29:
As is,  QGC displays battery status based on consumption, this is assuming that:

-user have correctly configured capacity 
-calculated in the reduced capacity on any older pack
-have correctly calibrated current (and voltage) sensor.
-all packs are about same age
-all packs are same capacity
-never flies a half-spent pack
-never land, reboot, then fly more. 

I've seen lots of crash logs & crashes due to poor routines with the method above, based on the assumption that 10Ah pack means you can safely fly 10Ah,  or even 8Ah.

Another (safer) approach, is to monitor for voltage.
Benefits are: 
-user may fly with very different pack age/health 
-a pack with a bad cell, will make voltage drop, and warn in time.
-flight with incorrect/half-spent battery (or reboot between flight) will still trigger alarm reliably.
- we may know, for a fact, than whenever a battery is at a given voltage, then we have x minutes to land.  so, a user with old/new/full/spent pack will still get the alarm at , let's say 14.4v . (4sx3,55v)  

Please make it a user selectable option in settings.

lakshaydang, 2017-04-21:
hi actually i am facing problem with battery percentage value. QGC showing correct voltage value but the percentage value is an issue. What is happening with me, if i raise the throttle the voltage value drops which is correct but at the same time the battery remaining percentage value increases instead of decreasing. Now what to do? 
Also this problem solve when i change the value of BAT_R_INTERNAL value from -1 to 0.016 but i don't know whether i am doing it right or not. Please help me? I have crashed my drone by more than 7 times now.

DonLakeFlyer, 2017-04-21:
@lakshaydang QGC just relays the information it gets from the vehicle. I'd ask for help from the developers that support the firmware you are using.

DonLakeFlyer, 2018-06-20:
QGC current uses information from firmware for battery display. Maybe an option to show voltage in toolbar instead of percent.

TrichoPilot, 2021-02-22:
How can I change the battery display in volts instead of percent in QGroundControl? I am using v4.1.1 (32 bit). Please help.

diegodab, 2021-08-16:
Does anyone know what to do for the app showing the real percentage value of battery remaining? It shows 100% at the moment of connection, and decreases proportionally to the real value from it( if it is charged at 80%, when it really has 40% it will show 50%).

AndKe, 2021-08-16:
@diegodab  That's normal when "percentage" is calculated by voltage, not actual capacity (by a BMS that knowns the state of charge)
for a 4S pack, there is a product named "BMSOne" that can act as an voltage based BMS, (by being calibrated for your battery pack - and one can work for many packs)  - otherwise, you would need proper BMS that keeps track of one pack, and you would need one for each pack.

diegodab, 2021-08-16:
@AndKe Thank you so much! Do you know a page where there are specified battery management systems for each type of battery? I am new to this and if you have it at hand it would save a lot of time.

AndKe, 2021-08-16:
@diegodab 
You have one that seems good, and very expensive (I did not test it, just observed it's existence, seems good, but overpriced.)
https://rotoye.com/batmon/BatMon-v4-03-2-packs-p273850043

Then you have the one I make for Solo (usable for many other things too, even if you should need to add extra power shuncts to exceed 60A)   - it's cheap and simple, https://madhacker.org/bmsone/

diegodab, 2021-08-16:
@AndKe Thank you again! My teammate / boss just told me that they already use a BSMOne, and that if they did not have it, it could not even read the voltage value to do the approximation. Is correct what he told me (he is not completely sure)? Do you know if the cause may be a parameter from the BSMOne or the actual QGC?  I just started the internship ;)

AndKe, 2021-08-16:
@diegodab I did not understood the statement: "and that if they did not have it, it could not even read the voltage value to do the approximation. "

BMSone does monitor individual cells, (and provide the pack voltage and current)  - being correctly set up (it uses a table for linearization), it will produce a nice, smooth SOC percentage that is reliable for any given battery type.
- it does not monitor charging, but a configured device will produce correct SoC when given a used battery to.  
Advantage of this method is that you do not one device per battery
(all is explained on the webpage)

diegodab, 2021-08-16:
@AndKe What I meant by that statement is that, according to my teammate, if the BSMOne was not connected, the QGC wouldn´t even know the decrease rate of the voltage to do the linear approximation, so it is working. 
The problem we have is that the battery´s capacity reading always starts at 100%, and does a linear approximation from that, but it never starts at the real value (if it ws at 10%, it would still show 100% when starting the app, and would show in this case 10% in QGC when the real value was 1%).

I have found that there are two modes to evaluate the SoC:

S0: default, voltage-based state of charge ("Should always use that mode"; from official information recommend to use this hahaha).
S1: capacity-based state of charge

Maybe I need to change it to S1.
Thank you so much for your help!

AndKe, 2021-08-16:
S1 requires you to always start out at a preprogrammed, known capacity (full useful capacity)  - but will then show real capacity based on consumed power.
S0 will start display higher-than-real SoC due to the load-less, higher cell voltage.
The SoC derived from voltage can only be precise at normal hover/cruise, (at typical load)  - before flight, the low load will always cause higher voltage, and so higher SoC than real.

diegodab, 2021-08-16:
Okay, thank you! As a last question, do you know how to manually change the way that battery remaining is shown? If I click on the percentage, I can see the Voltage, the mAh consumed, etc., and it would be great if I could see the voltage without clicking in anything :)

AndKe, 2021-08-16:
@diegodab  - I often use MAVProxy and a self-developed google Glass solution for flight monitoring, but this 4 year old issue is about just that, ... make QGC have an option to display voltage instead of percentage.

DonLakeFlyer, 2024-10-13:
Voltage in toolbar has been there for a while:
[image]

closed 2024-10-13 by DonLakeFlyer