mavlink/qgroundcontrol #4371: Compass calibration ArduPilot

lvale, 2016-12-31:
Latest QGC build on MacOS.

Onboard calibration doesn't appear as option

Only 2 compasses available (not 3)

[image]
[image]
[pixracer_compass_cal_apm343.mavlink.zip](https://github.com/mavlink/qgroundcontrol/files/679389/pixracer_compass_cal_apm343.mavlink.zip)

DonLakeFlyer, 2016-12-31:
There aren't options to do onboard or offboard. It's automatic based on what the vehicle supports. Let me try mine again.

DonLakeFlyer, 2016-12-31:
[image]

That looks like 3 compasses. The numbering is wrong though. Should be 1, 2, 3.

DonLakeFlyer, 2016-12-31:
Works for me. When you click Ok you get this:
[image]

No fancy images. Just tells you to rotate randomly.

lvale, 2016-12-31:
ok. Same thing here, except that mine doesn't show the 3rd compass :)

DonLakeFlyer, 2016-12-31:
Yeah I only saw two at first as well!

closed 2016-12-31 by lvale