mavlink/qgroundcontrol #4374: ArduPilot: Support new FRAME_CLASS, FRAME_TYPE airframe setup

DonLakeFlyer, 2016-12-31:


rmackay9, 2016-12-31:
For AC3.5 we are introducing a FRAME_CLASS parameter and renaming FRAME to FRAME_TYPE so it would be good to update the frame selection screen.

Possible values from FRAME_CLASS are 0:Undefined, 1:Quad, 2:Hexa, 3:Octa, 4:OctaQuad, 5:Y6, 6:Heli, 7:Tri, 8:SingleCopter, 9:CoaxCopter
Possible values for FRAME_TYPE are 0:Plus, 1:X, 2:V, 3:H, 4:V-Tail, 5:A-Tail, 10:Y6B

In case it helps, this is the equivalent screen I've mocked up for MP.  I'm sure QGroundControl will look very different but perhaps it gives a clue as to what it could look like.
[image]

Equivalent to-do for MP: https://github.com/ArduPilot/MissionPlanner/issues/1442

closed 2017-02-16 by DonLakeFlyer

OmahaCounsel, 2017-02-18:
This issue is NOT solved. There is no way to configure a y6b . The coax is the only choice. Even after loading the developer build.

DonLakeFlyer, 2017-02-18:
@OmahaCounsel Do you mean this configuration:
[image]

What OS are you using? Maybe the build drops aren't working correctly.

OmahaCounsel, 2017-02-21:
I have never see that screen.
My operating system is Windows 10 home. 64 bit running on a Samsung notebook.
I tried the qground control with both with and with out the px4 stack. Would only work with px4 stack but then no option for  a y6 b.

With respect,

Brent W. Nicholls, J.D., LL.M
Kasaby & Nicholls, LLC
300s. 19th Street #300
Omaha, NE 68102
402.502.0600 (p)
402.615.5926.(f)

[screen shot 2017-02-18 at 2 22 10 pm]<https://cloud.githubusercontent.com/assets/5876851/23097313/b01a9f90-f5e5-11e6-97ae-89a7f0eb0a0b.png>

DonLakeFlyer, 2017-02-21:
You ahve to be using a Daily build not a Stable build.

OmahaCounsel, 2017-02-21:
Don:

First let me belaboredly thank you for responding to my post.
At this point I have opened up the quad to rewire it to use the not as good tradition y6 type rotation but that mess with my props as I was going with larger bottom props to avoid the wash .
Perhaps I am doing something wrong in loading the software or firmware.  I say this because in fact I don’t even see the select frame type screen you sent me in your email.  The frame selection I see has several columns of pictures of frame types one of which (the only y6) is called y6 coaxial. In an effort to uncover the y6b option I uninstalled qgrond control and then reinstalled the lasted version selecting the px4 build. I still got nothing like the screen you are pointing me to, and in fact no apparent way to select frame type other than in the columns of pictures. I believe I down loaded the latest firmware, but perhaps not.

Any advice on how to get to that screen would be appreciated.

With respect,

KN LAW

Brent Nicholls, J.D., LL.M.
Brent@KNLAWpc.com<mailto:Brent@KNLAWpc.com>
www.KNLAWpc.com

300 S. 19th Street
Suite 300
Omaha, Nebraska 68102
402.502.0600
402.614.6540 (fax)

IRS CIRCULAR 230 DISCLOSURE: Unless expressly stated otherwise, any U.S. federal tax advice contained in this e-mail, including attachments, is not intended or written by KN Law, P.C., L.L.O. to be used, and any such tax advice cannot be used, for the purpose of avoiding penalties that may be imposed by the Internal Revenue Service.

CONFIDENTIAL COMMUNICATION: This electronic mail message contains confidential information which is (a) attorney – client privileged communication, work product, proprietary in nature, or otherwise protected by law from disclosure, and (b) intended only for the use of the Addressee(s) named herein. If you are not an Addressee, or the person responsible for delivering this to an Addressee, you are hereby notified that reading, copying, or distributing this message is prohibited. If you have received this electronic mail message in error, please reply to the sender and take the steps necessary to delete the message completely from your computer system.

From: Brent@knlawpc.com
Sent: Tuesday, February 21, 2017 2:06 AM
To: mavlink/qgroundcontrol
Cc: mavlink/qgroundcontrol; Mention
Subject: Re: [mavlink/qgroundcontrol] ArduPilot: Support new FRAME_CLASS, FRAME_TYPE airframe setup (#4374)

I have never see that screen.
My operating system is Windows 10 home. 64 bit running on a Samsung notebook.
I tried the qground control with both with and with out the px4 stack. Would only work with px4 stack but then no option for  a y6 b.
With respect,

Brent W. Nicholls, J.D., LL.M
Kasaby & Nicholls, LLC
300s. 19th Street #300
Omaha, NE 68102
402.502.0600 (p)
402.615.5926.(f)

OmahaCounsel, 2017-02-21:
Yes. I originally tried the stable build then the most recent or daily, as downloaded last Thursday I believe.

Again thank you for your help.

KN LAW

Brent Nicholls, J.D., LL.M.
Brent@KNLAWpc.com<mailto:Brent@KNLAWpc.com>
www.KNLAWpc.com

300 S. 19th Street
Suite 300
Omaha, Nebraska 68102
402.502.0600
402.614.6540 (fax)

IRS CIRCULAR 230 DISCLOSURE: Unless expressly stated otherwise, any U.S. federal tax advice contained in this e-mail, including attachments, is not intended or written by KN Law, P.C., L.L.O. to be used, and any such tax advice cannot be used, for the purpose of avoiding penalties that may be imposed by the Internal Revenue Service.

CONFIDENTIAL COMMUNICATION: This electronic mail message contains confidential information which is (a) attorney – client privileged communication, work product, proprietary in nature, or otherwise protected by law from disclosure, and (b) intended only for the use of the Addressee(s) named herein. If you are not an Addressee, or the person responsible for delivering this to an Addressee, you are hereby notified that reading, copying, or distributing this message is prohibited. If you have received this electronic mail message in error, please reply to the sender and take the steps necessary to delete the message completely from your computer system.

DonLakeFlyer, 2017-02-21:
Please send a screen shot of what you are seeing.

OmahaCounsel, 2017-02-21:
Ok will do I DO however  want to try downloading the right version one more time , perhaps from my office computer.
What would be the best version to use way to go about that so that I can get a y6b with a stable-is platform?
I think I have version 3.1.4 now

KN LAW

Brent Nicholls, J.D., LL.M.
Brent@KNLAWpc.com<mailto:Brent@KNLAWpc.com>
www.KNLAWpc.com

300 S. 19th Street
Suite 300
Omaha, Nebraska 68102
402.502.0600
402.614.6540 (fax)

IRS CIRCULAR 230 DISCLOSURE: Unless expressly stated otherwise, any U.S. federal tax advice contained in this e-mail, including attachments, is not intended or written by KN Law, P.C., L.L.O. to be used, and any such tax advice cannot be used, for the purpose of avoiding penalties that may be imposed by the Internal Revenue Service.

CONFIDENTIAL COMMUNICATION: This electronic mail message contains confidential information which is (a) attorney – client privileged communication, work product, proprietary in nature, or otherwise protected by law from disclosure, and (b) intended only for the use of the Addressee(s) named herein. If you are not an Addressee, or the person responsible for delivering this to an Addressee, you are hereby notified that reading, copying, or distributing this message is prohibited. If you have received this electronic mail message in error, please reply to the sender and take the steps necessary to delete the message completely from your computer system.

DonLakeFlyer, 2017-02-22:
If you are using anything other than ArduPilot Copter 3.5 builds then Stable is fine.

OmahaCounsel, 2017-02-22:
thanks

KN LAW

Brent Nicholls, J.D., LL.M.
Brent@KNLAWpc.com<mailto:Brent@KNLAWpc.com>
www.KNLAWpc.com

300 S. 19th Street
Suite 300
Omaha, Nebraska 68102
402.502.0600
402.614.6540 (fax)

IRS CIRCULAR 230 DISCLOSURE: Unless expressly stated otherwise, any U.S. federal tax advice contained in this e-mail, including attachments, is not intended or written by KN Law, P.C., L.L.O. to be used, and any such tax advice cannot be used, for the purpose of avoiding penalties that may be imposed by the Internal Revenue Service.

CONFIDENTIAL COMMUNICATION: This electronic mail message contains confidential information which is (a) attorney – client privileged communication, work product, proprietary in nature, or otherwise protected by law from disclosure, and (b) intended only for the use of the Addressee(s) named herein. If you are not an Addressee, or the person responsible for delivering this to an Addressee, you are hereby notified that reading, copying, or distributing this message is prohibited. If you have received this electronic mail message in error, please reply to the sender and take the steps necessary to delete the message completely from your computer system.

OmahaCounsel, 2017-02-26:
Well I never get to the setup screen you are looking at . Here is what I see after reinstalling q-ground control and updating firmware.

[image1.JPG]
With respect,

[image2.JPG]
See no choice for y6b. .???

Brent W. Nicholls, J.D., LL.M
Kasaby & Nicholls, LLC
300s. 19th Street #300
Omaha, NE 68102
402.502.0600 (p)
402.615.5926.(f)

DonLakeFlyer, 2017-02-26:
No images. You can't post images from image. Need to do it from GitHub.

OmahaCounsel, 2017-02-26:
Ok I'm giving up there just is no way to fly a y6b with this program . On option to chose y6b exist if you used the px4 stack

With respect,

Brent

Kasaby & Nicholls
300 S. 19th Street #300
Omaha, NE 68102
402.502.0600

DonLakeFlyer, 2017-02-26:
I'm sorry but I have no idea what screen you are not seeing the Y6b on since your images did not make it through. The settings for frames is just a parameter so it is absolutely possible.

OmahaCounsel, 2017-02-26:
I tried to send you a picture of the screen been trying to over come this for three weeks . Downloaded and reinstalled q ground control 8 times .chosing  px4 pro the frame type screen only show y6 coaxial the us a drop down select but that only gives you the one option y6 coxial.

With respect,

Brent

Kasaby & Nicholls
300 S. 19th Street #300
Omaha, NE 68102
402.502.0600