mavlink/qgroundcontrol #4392: Multi-vehicle: disconnect from selected vehicle, selector disappears

sanderux, 2017-01-04:
When connected to 2 vehicles (A and B) and having vehicle A active.
if vehicle A disconnects the vehicle selector disappears and there is no way to switch to vehicle B

DonLakeFlyer, 2017-01-04:
Are you running a daily build? Should already be fixed.

DonLakeFlyer, 2017-01-04:
Wait this is different. When Vehicle A disconnects it will automatically switch to Vehicle B and the selector will go away because there is only one vehicle. Just tested this and it is doing the right thing?

sanderux, 2017-01-04:
`when it looses connection it doesnt. It shows the connection lost screen with a disconnect button.

sanderux, 2017-01-04:
It might be an older version, its a daily but not todays :)

DonLakeFlyer, 2017-01-04:
Should have been fixed on Nov 30: https://github.com/mavlink/qgroundcontrol/pull/4265

sanderux, 2017-01-04:
I'll check, maybe my version is old. Sorry

closed 2017-01-04 by sanderux

DonLakeFlyer, 2017-01-04:
Daily builds will have date in title bar.