mavlink/qgroundcontrol #4406: iOS: QGC crashed when connecting through wifi

CarlOlsson, 2017-01-08:
After updating to the latest QGC for iOS the app crashes as soon as you connect a vehicle to the device over wifi.
I have tried with an iphone 6 and an ipad 4 both running iOS 10

DonLakeFlyer, 2017-01-08:
Thanks for the report. Let me take a look.

DonLakeFlyer, 2017-01-08:
Damn, I can repro as well.

dogmaphobic, 2017-01-08:
It runs here. I assume this is the "release" version (not the TestFlight)?

DonLakeFlyer, 2017-01-08:
I found it.

dogmaphobic, 2017-01-08:
Where?

DonLakeFlyer, 2017-01-08:
Vehicle::_updatePriorityLink is fully #ifdef'ed out on iOS due to NO_SERIAL_LINK. But only the serial link part of that should be out. Fallback to first link should still be in.

dogmaphobic, 2017-01-08:
Are you fixing it or should I? How to add it back to Stable_V3.1?

DonLakeFlyer, 2017-01-08:
Once master goes through I'll move it to Stable_V3.1 and I'll move the tag and you'll need to drop Stable and TestFlight again.

dogmaphobic, 2017-01-08:
Yeah, that came in with #4349. That whole multiple link is voodoo to me :P

DonLakeFlyer, 2017-01-08:
Yeah I knew I'd break something with that one. Way too much change to come clean.

CarlOlsson, 2017-01-09:
Awesome now it works, thanks!

closed 2017-01-09 by CarlOlsson

DonLakeFlyer, 2017-01-09:
Thanks for the quick bug report @CarlOlsson