mavlink/qgroundcontrol #4411: UVC implementation has issue with some UVC devices

cda725, 2017-01-09:
I have discovered a slight issue that needs addressed and hope those with more insight into the matter agree and know how to achieve what I am about to explain - hoping it is not a QT / QT version issue. I have tried remedy but I believe this is slightly above my skill.

Some UVC devices are accessed by Android as USB devices and don’t list under /dev/video (or other camera type device); instead they are found (at least on all my Androids) at /dev/bus/usb/001/00n (n being any number) and QGC will not make use of them. It is so odd that almost EVERY android camera app has no issue with ANY UVC card I have tried.

This behavior is even more perplexing because in Linux, Mac and Windows ANY UVC (and even driver based capture cards) of this type are detected by QGC without issue as a video/camera device and work flawlessly. Again, CameraFI and other camera apps for Android have no issue detecting USB UVC capture devices that QGC is simply not seeing so I am assuming there is some handler to force grabbers to be detected as cameras; something that is apparently built-in to Linux, Windows and Mac somehow and not present in Android. 

Being that CameraFi and others can see the UVC grabber type devices that QGC can not I am assuming this is not too difficult but I am not sure and am desperately seeking some wisdom here.

The call to the camera is quite simple in QGC...

`
QList<QCameraInfo> cameras = QCameraInfo::availableCameras();
    foreach (const QCameraInfo &cameraInfo, cameras) {
        if(cameraInfo.description() == vSource) {
            _videoSourceID = cameraInfo.deviceName();
            emit videoSourceIDChanged();
            qCDebug(VideoManagerLog) << "Found USB source:" << _videoSourceID << " Name:" << _videoSource;
            break;
        }
    }
`

But this simply will NOT call any capture device that is not under /dev/video or /dev/camera...

It is like QT is relying on a depreciated method of detecting video input devices.

Could a function be implemented that would treat UVC driverless capture cards as cameras since they are technically a video device? 

Is there an easy way handle this? I am willing to attempt to code it but I will need a little guidance regarding a start point as I am not familiar enough with programmatically communicating with hardware through QT and Android.

dogmaphobic, 2017-01-09:
This would have to be dealt with within Qt itself. There isn't much you can do from QGC unless you want to write your own (Android) camera handler. Qt uses direct, Android (Java) calls to interface its multimedia API. If you look in `qtmultimedia/src/plugins/android/src/wrappers/jni/androidcamera.cpp` you will see functions like:

```
int AndroidCamera::getNumberOfCameras()
{
    return QJNIObjectPrivate::callStaticMethod<jint>("android/hardware/Camera",
                                                     "getNumberOfCameras");
}
```

And so on... That's where Qt iterates and identifies the available *cameras* within Android.

dogmaphobic, 2017-01-09:
Looking at the Android documentation (https://developer.android.com/reference/android/hardware/Camera.html) it seems Qt is using an older API to access the camera. You would need to bring this up with the Qt dev team and see what they have to say.

cda725, 2017-01-09:
Thank you!!! I was afraid it was a QT issue - Ugh. Off to cal the powers that be.

cda725, 2017-01-09:
Just curious - is it that they are using android.hardware.camera?

dogmaphobic, 2017-01-09:
I can't quite answer that as I know close to nothing about Android. I looked up how Qt was iterating cameras and found the code using android/hardware/camera. When I went to the Android website to see what that interface provided, I noticed the comment up top saying that was a deprecated API and people should use camera2 instead. That's as far as I went. I do not know what either interface does or is capable of.

DonLakeFlyer, 2017-01-09:
Maybe different in 5.8?

dogmaphobic, 2017-01-09:
Looking at the sources, it doesn't seem anything changed. It's still calling `android/hardware/Camera`. If there has been some change elsewhere I don't know and I won't be able to test as I don't have any hardware that shows the problem listed in this issue.

cda725, 2017-01-09:
What do you need? I have extra Android tablets - Do you still have the UVC device?

dogmaphobic, 2017-01-09:
I have one single UVC device. An "easyCap" USB dongle, which works fine.

cda725, 2017-01-09:
In Android? If so please send me the amazon link... I will just use those instead...

dogmaphobic, 2017-01-09:
My bad. I've been working on so many different things at once that I lost track of what is what. The (Android) tablet I just tested is rooted and has all sorts of changes done to it. I tried on a "normal" tablet (Nexus 7, Android 6.x) and it doesn't work there. While looking at all this, I started to remember all the crazy stuff I had to do to get Android to see the UVC device.

The Android documentation for the new `camera2` interface has explicit documentation for external cameras. Given that Qt does not use it (it still uses the old `camera` interface), that won't help.

https://source.android.com/devices/camera/versioning.html#module_version

*External camera (e.g. USB hot-plug camera) support. The API updates specify the camera static info is available only when camera is connected and ready to use for external hot-plug cameras. Calls to get static info will be invalid calls when camera status is not CAMERA_DEVICE_STATUS_PRESENT. The framework counts solely on device status change callbacks to manage the available external camera list.*

In summary, it can be done but it requires changes. Here is something I found that could lead to some (non-rooted) solution while waiting for Qt to update:

https://github.com/saki4510t/UVCCamera

cda725, 2017-01-10:
Thank you... I will look into that. I wrote QT about the issue as well to see if they may also have suggestions.

cda725, 2017-01-11:
Hi everyone! 

The issue has been raised with QT - https://bugreports.qt.io/browse/QTBUG-58026 - I am not sure what it means but yesterday they assigned it a "P3 - somewhat important" priority level. 

I looked at the Android workaround but until either QT resolves the issue or there is another fix I may have to stay with either  windows tablets or Jaguar Board X86 (Pi like) devices running windows or linux housed in a custom shell with a touch screen. 

Since we know this is a QT problem should I close this issue here or leave it open until a solution is found? Perhaps leaving it open here will allow others with ideas to offer input until a final fix is implemented.

DonLakeFlyer, 2017-01-12:
Leave it open so we can keep checking Qt for the fix.

dogmaphobic, 2018-06-01:
This came up again today. As of today, with Qt 5.11, Qt is still using a deprecated Android Camera API, which does not support external cameras: 

http://code.qt.io/cgit/qt/qtmultimedia.git/tree/src/plugins/android/src/wrappers/jni/androidcamera.cpp?h=5.11.1#n768

DonLakeFlyer, 2018-06-20:
Closing since we are stuck with what Qt gives us

closed 2018-06-20 by DonLakeFlyer