mavlink/qgroundcontrol #4412: HDOP/VDOP values not updated in QGC MainToolBar: GPS Status

samacumen, 2017-01-09:
I would like to bring to your notice that the hdop/vdop is not getting updated in QGC in MainToolBar display
I checked that in the stable branch v.3.1.1 the GPS based changes were moved from MainToolBar.qml to MainToolBarIndicators.qml, and also some re-arrangement and code clean up has occured.

Tried to simulate this using normal flight and in HIL mode, did not seem to update.

Please see the image below:

[image]

DonLakeFlyer, 2017-01-09:
Tested using a real vehicle with both 3.1.1 and master and it's working fine. If you are getting 0's then most likely the vehicle is not sending the values across. Check using mavlink inspector (Widgets/mavlink inspector). You should see this:
[image]

hdop and vdop are in the eph and epv values.

samacumen, 2017-01-10:
Thanks @DonLakeFlyer for testing this. I did not seem to have the latest firmware for the pixhawk. It works now, you may close the issue.

closed 2017-01-10 by samacumen