mavlink/qgroundcontrol #4413: iOS: Can't change min voltage

CarlOlsson, 2017-01-09:
In the power tab you can't change the min voltage of the battery cells on iOS. If you try to change it and then switch tab, e.g. to sensors, and then switch back again the voltage level is reset always to 3.4V.

It works however on QGC for Ubuntu.

DonLakeFlyer, 2017-01-10:
Did you hit the done button on the virtual keyboard before changing pages?

CarlOlsson, 2017-01-10:
Yes, just tried it again and it is not possible to change the voltage

DonLakeFlyer, 2017-01-12:
Does it work if you change BAT_V_EMPTY from the parameter editor?

DonLakeFlyer, 2017-01-12:
I tried to repro this on both master and stable. Both worked fine for me. Can you detail out click by click what you are doing?

CarlOlsson, 2017-01-25:
Hi Don,

Tried it again now.
1) Pressed power tab
2) Pressed battery voltage field
3) Virtual keyboard shows up, changed voltage from 3.4 to 3.3
4) Pressed enter on virtual keyboard and it disappeared 
5) Voltage is still 3.3 volt
6) Pressed sensor tab
7) Pressed power tab -> Voltage is again reset to 3.4V

However, if I change the parameter in the parameter tab it works and the voltage in displayed correctly in the power tab

DonLakeFlyer, 2017-05-24:
Are you talking about the "Empty Voltage (per cell)" field? Also when you say pressed enter on the virtual keyboard, do you mean pressed "Done"?

It's all still working fine for me

DonLakeFlyer, 2017-12-31:
I just tried this again and it is still working fine for me

closed 2017-12-31 by DonLakeFlyer