mavlink/qgroundcontrol #4414: Survey functionality, creat auto take off and landing/RTL options

leondwyer, 2017-01-10:
One of the great thing MS does is it handles the auto take of land / rtl waypoint for you to in the survey planner.

It saves having to set them up yourself. there just simple check boxes on the options screen.

DonLakeFlyer, 2017-01-11:
Also need to consider flight speed since I didn't add that in my referenced pull. The reason is that with QGC a Survey item is slightly different than in Mission Planner. In Mission Planner the Survey widget sets up an entire mission. Whereas in QGC a Survey is only considered a part of a survey and you can have multiple surveys in a mission for example. Need to rationalize that difference with this request.

DonLakeFlyer, 2017-04-13:
RTL end action and flight speed changes are now handled in first mission settings item. So this is all automatic now.

closed 2017-04-13 by DonLakeFlyer