mavlink/qgroundcontrol #4416: Increase Geotag Accuracy

saitej09, 2017-01-10:
Hi I have tried using the ExifParser. There is a loss in accuracy of about 100 cm in conversion from decimal lat/lon to dms and writing to imagebuffer.

Can you explain how to increase the accuracy?

Relevant code:

`
 gpsData.readable.extendedData.gpsLat[0] = abs(static_cast<int>(coordinate.latitude()));
    gpsData.readable.extendedData.gpsLat[1] = 1;
    gpsData.readable.extendedData.gpsLat[2] = static_cast<int>((fabs(coordinate.latitude()) - floor(fabs(coordinate.latitude()))) * 60000.0);//gives lat min
    gpsData.readable.extendedData.gpsLat[3] = 1000;
    gpsData.readable.extendedData.gpsLat[4] = 0;
    gpsData.readable.extendedData.gpsLat[5] = 1;

    gpsData.readable.extendedData.gpsLon[0] = abs(static_cast<int>(coordinate.longitude()));
    gpsData.readable.extendedData.gpsLon[1] = 1;
    gpsData.readable.extendedData.gpsLon[2] = static_cast<int>((fabs(coordinate.longitude()) - floor(fabs(coordinate.longitude()))) * 60000.0);
    gpsData.readable.extendedData.gpsLon[3] = 1000;
    gpsData.readable.extendedData.gpsLon[4] = 0;
    gpsData.readable.extendedData.gpsLon[5] = 1;
`

Thanks!!

DonLakeFlyer, 2017-01-10:
@birchera Andreas?

birchera, 2017-01-11:
The current rational format is (ddd/1,mmmm/1000,0/1) and has an accuracy in the order of <2m

If we want to increase accuracy we could go for (ddd/1,mm/1,ssss/1000). This would bring us to sub cm accuracy. Such accuracy is only beneficial if you have a PPK/RTK device.

@saitej09 what are your requirements?

saitej09, 2017-01-11:
@birchera 

Hi ..
We have the exact same requirements. RTK/PPK device

birchera, 2017-01-11:
I can propose a patch. This week will be difficult, but next week should be possible.

saitej09, 2017-01-11:
@birchera 

Sure!! Thanks. It will be great if u can give some hints and I too can try from my side.

saitej09, 2017-01-18:
@birchera  any update on the patch ?
Thanks!

PR #4460 (2017-01-31): Increase accuracy of geotags
Fixes #4416.

* Old format for rational was (ddd/1,mmmm/1000,0/1), new (ddd/1,mm/1,ssss/1000).
* This brings accuracy of representation to sub cm.

@saitej09 can you test this?
closed 2017-01-31 by DonLakeFlyer