mavlink/qgroundcontrol #4417: Command long messages need to handle the confirmation field

dogmaphobic, 2017-01-10:
Adding here before it falls through the cracks.

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1911

The `confirmation` field is used to identify which "instance" of the command is being processed. That is, when you send a given *COMMAND_LONG* message, the *ACK* message will have the value set in the original `confirmation` field. For example, if you send two or more (same) commands in a row, you will need to be able to identify for which command an ACK is being received.

This is based on a conversation with @LorenzMeier. I'm not sure where within QGC there could be a case where this is needed.

DonLakeFlyer, 2017-01-10:
It's going to be a pain to support this. The reason is that no firmware supports this now. Hence QGC has no idea how it is going to get acks back. Sometimes it may get a confirmation back that matches sometimes it won't. Having to case out by firmware version for all these features gets out of hand really fast. It's going to need a capability bit or something to let the groundstation knows what can happen.

dogmaphobic, 2017-01-10:
Is there any (unique) command you can think of that's sent more than once in a row? That is, will this ever be set to 1 or more? This came up when @bkueng noticed a very long stream of MAV_CMD_PREFLIGHT_CALIBRATION being sent clobbering the queue.

DonLakeFlyer, 2017-01-10:
> a very long stream of MAV_CMD_PREFLIGHT_CALIBRATION being sent clobbering the queue.

That would be a bug. QGC should only retry five times. Also newest PX4 firmwares correctly acks this commands. So if using new firmware you should get no retries. If you are then that's a firmware bug.

Not sure what you mean by a unique command sent more than once in a row? They all will be sent more than once it they don't get an ack. But only by the retry logic. Nothing gets sent more than once to start with.

dogmaphobic, 2017-01-10:
I have to see what's up with MAV_CMD_PREFLIGHT_CALIBRATION. I have not tried it myself yet. This is still a bit confusing to me.

Forget the "unique" part. Only now I see that the ACK only tells you which command it is responding, without its arguments. Therefore, there is no way for you to know. For example, if you were to send two MAV_CMD_PREFLIGHT_CALIBRATION commands, one for gyro and another for mag, you would receive an ACK for MAV_CMD_PREFLIGHT_CALIBRATION but you would not know if it was for the gyro or the mag calibration command (if you had sent them both in a row).

DonLakeFlyer, 2017-01-10:
That is correct. Except that QGC only ever has one COMMAND_LONG out to the vehicle at a time. Additional ones get queued up until the previous one completes.

dogmaphobic, 2017-01-10:
> Additional ones get queued up until the previous one completes.

That answers my main question (this issue). If that's the case, QGC will never run into the "whose ACK is this for" situation.

closed 2017-01-10 by dogmaphobic