mavlink/qgroundcontrol #4425: On Start QGC just hangs.

tubeme, 2017-01-11:
Still cannot figure out the reason for this. On start up QGC just does not start. The mouse pointer starts rolling for a second and no window opens. 

I tried 3.1.0 ; 3.1.1 and the current Master as well. All are the same. I had a copy of a 20Dec2016 master so I installed it and it works!!!.

Might be related to Intel Graphics. The PC is with embedded graphics but with the default microsoft drivers. 

On all versions I tried I used all the 3 modes of start. Neither one worked.

I tried to execute the QGC in the shell (terminal). It did nothing. No message no start.

Tried to execute it as Admin again no success...

NaterGator, 2017-01-11:
@tubeme there should be 3 shortcuts in your start menu that force progressively degraded GPU acceleration (down to full software rendering). Can you try them to determine if the Intel iGPU is indeed at fault?

tubeme, 2017-01-11:
@NaterGator Exactly what I did. Tried all the 3 options to each install 3.1 3.1.1 and todays Master.

Only my 20 Dec 2016 master works.

tubeme, 2017-01-11:
Forgot to mention it is Windows 10 with latest updates.

tubeme, 2017-01-12:
I upgraded the Intel drivers. It is Intel HD 5500. Now with the good drivers nothing again. Does not start. Should be something else. May be some other device or sth... Is there a LOG??

DonLakeFlyer, 2017-01-12:
Is it hanging or just not starting. Do you have to kill the QGC process?

DonLakeFlyer, 2017-01-12:
@NaterGator Could you work with Vasil to try to track this down? Maybe get him a debug build so he can get console output. Also would there be a way to get a windows build to output console to command prompt through some sort of command line option? I'm in the middle of a ton of stuff right now so will take me a while to help.

bwmcclai, 2017-01-13:
I'll just chime in to say that I am experiencing the exact same problem.  

- Windows 10
- Tried multiple builds
- Tried all compatibility launch modes

Mouse spins for a second and then nothing....

DonLakeFlyer, 2017-01-13:
Can you try manually cleaning out QGroundControl:
- Uninstall QGC
- Make sure there is no C:\Program files (x86)\qgroundcontrol directory
- Install stable
- Run QGC from the command line with this "qgroundcontrol.exe --clear-settings"

bwmcclai, 2017-01-13:
@DonLakeFlyer I followed your steps exactly.  for "Install Stable" I used this release: https://github.com/mavlink/qgroundcontrol/releases/tag/v3.1.0

When I try to run it from the command line, nothing happens, just like when trying normally.

By the way, 3.0.2 works for me when I set it to run in GPU compatibility mode.

NaterGator, 2017-01-13:
@DonLakeFlyer  I'm pretty slammed during the week but I can help with this over the weekend. I'll see if I can repro quickly on my Win10 work machine, if so I can probably track down the bug in a few minutes.

DonLakeFlyer, 2017-01-13:
FYI: It doesn't repro for me on my Win10 VM.

bwmcclai, 2017-01-13:
Thanks for the quick replies guys.  I appreciate it.

NaterGator, 2017-01-13:
No repro here on my work machine. For those of you having this issue, can you run [DebugView++](https://debugviewpp.wordpress.com/2014/01/04/7/) to capture a log while trying to start QGC?

DonLakeFlyer, 2017-01-13:
@NaterGator Run with "qgroundcontrol --logging:full" and DebugView++? That will capture QGC output?

NaterGator, 2017-01-13:
Yes, the default QtMessageHandler invokes [OutputDebugString](https://msdn.microsoft.com/en-us/library/windows/desktop/aa363362(v=vs.85).aspx) which DebugView++ captures. Anything that shows up in the QGC debug console should be caught by DebugView++.

bwmcclai, 2017-01-13:
[13036] Settings location "C:/Users/Brian.McClain/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true
[13036] Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n*Log.debug=true\n"
[13036] MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/Brian.McClain/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs"
[13036] JoystickLog: Available joysticks
[13036] JoystickLog:   "vJoy Device" axes: -1 buttons: -1 hats: -1 isGC: false

NaterGator, 2017-01-13:
If you have a joystick plugged in, can you unplug it and try again?

bwmcclai, 2017-01-13:
I don't have one plugged in.

NaterGator, 2017-01-13:
Hmm, doing a little code review I'm pretty sure I've got it. One sec and I'll submit a PR for a build for you to try.

NaterGator, 2017-01-13:
Please try this build and let me know if it works: https://ci.appveyor.com/api/buildjobs/p5mdclv17gl5mvo0/artifacts/QGroundControl-installer.exe

vsobalski, 2017-01-13:
Hi,
I just had the same issue, and ended up running into this thread. I have Windows 10, and had issues with the version 3.11 and 3.10 of QGroundControl. The packaged you submitted seemed to have corrected the problem on my computer. Thanks a lot

bwmcclai, 2017-01-13:
works like a charm now.  THANKS!

closed 2017-01-13 by DonLakeFlyer

tubeme, 2017-01-14:
Thanks @NaterGator a lot. Now the qgroundcontrol starts perfect.

But there is another problem. The Joystick gets recognized but when I go to calibrate it, it states:

"Detected 0 joystick axess. To operate PX4 you need at least 4 axes."

and it does not allow me to calibrate the joystick. From the above posts I understand this is related to the joystick or other input devices. I have three of those - Keyboard with mouse pad, a trackball and joystick.

DonLakeFlyer, 2017-01-14:
@tubeme Can you move this to a new Issue?

neoalejo, 2019-03-12:
> Please try this build and let me know if it works: https://ci.appveyor.com/api/buildjobs/p5mdclv17gl5mvo0/artifacts/QGroundControl-installer.exe

I am having this same issue that apparently this build was able to solve. How can I access this build? Apparently the link is broken.