mavlink/qgroundcontrol #4428: Add a way to install PX4 LPE firmware version for px4fmu-v2

jgoppert, 2017-01-12:
See issue here: https://github.com/PX4/Firmware/pull/6307

dagar, 2017-01-12:
@DonLakeFlyer for now what do you think about putting this in the advanced drop down if the board is px4fmu-v2? Longer term we can do something better with the json firmware listing.

jgoppert, 2017-01-13:
If it helps, all px4 firmware versions are names  OS-BOARD-LABEL

We are stripping off all OS because it is always nuttx current.
The board is px4fmu-v2 etc.
The label identifies what set of modules were built. So far we have only been using default.

px4fmu-v2 is the only board that has both default and lpe labels. So the gui could have a label selectors at some point, but for now a drop down list of all available firmware would work.

closed 2026-10-05 by github-actions[bot]