mavlink/qgroundcontrol #4430: Joystick: On command the control surfaces shake.

tubeme, 2017-01-12:
Hello guys,

5-6 month ago I had a very successfull experiments with controling airplane with joystick.

Now after our new ground station is ready we put QGC 3.0.2 and the things with setting up the joystick got way better than before. But after the calibration the problem i have is there is no really commands transl;ating from the joystick to the control surfaces. They move but rather shake. Not possible to fly it this way.

Here is a video:

https://youtu.be/4bbvSPJoVvc

Any ideas?

dagar, 2017-01-12:
Do you have RC connected as well? What's COM_RC_IN_MODE set to?
Can you share the log?

Your ground control station is insane.

DonLakeFlyer, 2017-01-12:
Also 3.0.2 is pretty old now. I'd go with Stable 3.1. That is one hell of a groundstation!

DonLakeFlyer, 2017-01-12:
Wait you're not on 3.1 because of #4425

DonLakeFlyer, 2017-01-12:
@tubeme Do you have the ability to build your own QGC?

tubeme, 2017-01-12:
@dagar  The RC is conected, but as far as I remember if I don't touch the RC then the Joystick gets over.
The COM_RC_IN_MODE = 2. I will get the LOG tomorrow.

BTW If I could lay with my ground station I would do it :)

I designed it, a friend of mine produced it. We plan on commercial production. This is the first prototype.

[image]

[image]

@DonLakeFlyer  Exactly because we cannot run 3.1 because of #4425

I don't have the setup to compile my own version.

tubeme, 2017-01-12:
Forgot to mention. The ground station has triple redundancy on the input metod - Touch screen, Professional Trackball, Keyboard with pad.

And it has triple redundancy on the power supply. Internal 12 volt for 2 hours, can be connected to external 12v batery and 220v AC.

DonLakeFlyer, 2017-01-13:
Need to get a log file to the firmware folks to track down what is driving the servos.

DonLakeFlyer, 2017-01-13:
Can you try manually cleaning out QGroundControl:
- Uninstall QGC
- Make sure there is no C:\Program files (x86)\qgroundcontrol directory
- Install stable
- Run QGC from the command line with this "qgroundcontrol.exe --clear-settings"

dagar, 2017-01-13:
@tubeme right now you can't have joystick with QGC and RC connected simultaneously. They're fighting each other.
We've been talking about how to fix it on the firmware side. At a minimum either explicitly switching to one or the other via param, or a priority system.

tubeme, 2017-01-13:
Here is the LOG file:

http://logs.uaventure.com/view/kQS52VABfcUkHZYQNF4wU3

DonLakeFlyer, 2017-01-13:
> right now you can't have joystick with QGC and RC connected simultaneously. They're fighting each other.

@tubeme Is this the problem?

tubeme, 2017-01-14:
As far as I remember there was huge discussion about how it should be implemented and it was pretty long time ago. I thought that it is implemented such that the RC has a priority over the Joystick but if we don't touch the RC and have input from the joystick it performs the control. That is the most logical way of operation. It is very similar to Trainer Student relation with two RCs. In our case the Student is the Joystick the Trainer is the RC. Also it should have some form of manual switch between the two controls in the QGC interface.

So is this the problem?? They are fighting each other?

tubeme, 2017-01-14:
@dagar It is not a good idea to have either one. In real life situations they both are backup to each other. We use the joystick for long missions, but take off and land with the RC when we have connection. The Joystick is operated by UART so 10 times more range. If there is problem to one of the controls we have to make sure the other one is active right away.

dagar, 2017-01-14:
If we get priorities working both could be connected simultaneously. You'd select which one has priority with a param. Other than that I don't know how you could have them both working at the same time. Even hands off they're still sending the stick position constantly.

DonLakeFlyer, 2017-01-14:
There is an Issue in already on the whole "joystick takes control" support in QGC. More like an RC trainer type setup. But not work on it yet.

tubeme, 2017-01-14:
I think it is very well organized in OpenTX regarding the trainer port operation.
The easiest way I think is not with parameter, but with a button on the QGC interface. Klick the button you have RC click it once more you get Joystick. Just a software switch.

dagar, 2017-01-15:
Closing this because support needs to be added on the firmware side. Once we have that in place we can think about adding a mechanism to display and change the active controller in QGC. We could have both an RC switch and mavlink message, as well as handling failover.

https://github.com/PX4/Firmware/issues/5806

closed 2017-01-15 by dagar