mavlink/qgroundcontrol #4433: Adding support for px4fmu-v4pro

klopezal, 2017-01-13:
Hello, could you add support for the upcoming px4fmu-v4pro? So we can get the very nice automatic connection feature.  New PID/VID as in Nuttx config are : USB\VID_26AC&PID_0013. Thanks!

DonLakeFlyer, 2017-01-13:
Does this run the exact same firmware as other V4 boards? Or is there a different set of firmware builds for it?

DonLakeFlyer, 2017-01-13:
Does the bootloader return this for board id:
    static const int boardIDPX4FMUV4 = 11;   ///< PX4 V4 board, as from USB PID

klopezal, 2017-01-15:
No it is a different firmware (the processor is STM32F469), the target is px4fmu-v4pro. Board ID is  [13](https://github.com/PX4/Bootloader/blob/master/hw_config.h#L229), but is different from USB PID (0x0013)!

closed 2024-06-02 by HTRamsey