mavlink/qgroundcontrol #4440: PX4 pixhawk firmware flash based on reported size

dagar, 2017-01-14:
Binaries for the original pixhawk are limited to 1 MB. The pixhawk 2 is binary compatible with the pixhawk 1, but able to use the full 2 MBs of flash. Newer pixhawk 1s (http://store.mrobotics.io/category-s/113.htm) are also able to use the full 2 MBs.

All of these have the same board id, however the bootloader reports the flash size.

Pixhawk 1 - Flash px4fmu-v2_default
https://s3.amazonaws.com/px4-travis/Firmware/master/px4fmu-v2_default.px4

    Connected to bootloader:
    Version: 5
    Board ID: 9
    Flash size: 2080768
    Downloading firmware...

Pixhawk 2 (and newer Pixhawk 1) - Flash px4fmu-v**3**_default
https://s3.amazonaws.com/px4-travis/Firmware/master/px4fmu-v3_default.px4

    Connected to bootloader:
    Version: 5
    Board ID: 9
    Flash size: 1032192

DonLakeFlyer, 2017-01-14:
I assume you have those size reports reversed?

dagar, 2017-01-14:
No, looks fine to me :wink: 
(fixed it)

DonLakeFlyer, 2017-06-09:
Aren't there variants of these boards out there that have larger flash size but have silicon problem such that is you flash above first half they crash?

DonLakeFlyer, 2017-06-09:
@proficnc ?

I have a pull ready to go, but not sure about my above question.

DonLakeFlyer, 2017-06-09:
For example: I have a 3dr pixhawk that is pretty old. Yet it reports as a V2 with a large flash size.

DonLakeFlyer, 2017-06-09:
And if I remember correcrtly this was a board that had the flaky flash limit problems.

dagar, 2017-06-09:
My understanding is the bootloader should be correctly reporting the usable flash based on the particular revision stm32f4 revision. @davids5

DonLakeFlyer, 2017-06-09:
That would be a new bootloader though wouldn't it. Older board won't have a newer/smarter bootloader.

davids5, 2017-06-09:
@DonLakeFlyer, @dagar V5 of the bootloader checks for the version of silicon. Based on the silicon errata and reports the size appropriately

DonLakeFlyer, 2017-06-10:
Ok, so I need to check the bootloader version as well. What should I do if I see a V2 board which large flash space but an old bootloader? Flash V2 instead of V3?

davids5, 2017-06-10:
Yes. Start with a V2 build.  Then do a `ver all` that will tell you if it has the bug. If not V3 will work.

DonLakeFlyer, 2017-06-10:
I'm not sure what that means with respect to QGC flashing firmware. It talks directly to the bootloader over a serial link. There is no "ver all" or anything like that.

davids5, 2017-06-10:
Oh I missed the context. I was telling you how one could tell.  QGC can check the version of the bootloader.  If it is V5 (or greater) believe the size and use the correct FW V2 for 1 ~Mib and V3 for 2Mib. If it is not V5 or greater use V2.

DonLakeFlyer, 2017-06-10:
Got it, thanks.

closed 2017-06-20 by DonLakeFlyer