mavlink/qgroundcontrol #4442: Joystick: Cannot be calibrated.

tubeme, 2017-01-14:
Fixing #4425 probably created new problem with the joystick

The Joystick gets recognized but when I go to calibrate it, it states:

"Detected 0 joystick axess. To operate PX4 you need at least 4 axes."

and it does not allow me to calibrate. thus I cannot use it.

qgc is master past 3.1.1. with the #4425 fix.

DonLakeFlyer, 2017-01-14:
@NaterGator Must be a windows thing maybe. Working fine on OSX with my Logitech.

DonLakeFlyer, 2017-01-14:
@rjehangir Are you seeing similar problems with Stable?

DonLakeFlyer, 2017-01-14:
@bwmcclai Reported this in #4425:
[13036] Settings location "C:/Users/Brian.McClain/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true
[13036] Filter rules "qt.network.ssl.warning=false\nLog.debug=false\nLog.debug=true\n"
[13036] MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/Brian.McClain/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs"
[13036] JoystickLog: Available joysticks
[13036] JoystickLog: "vJoy Device" axes: -1 buttons: -1 hats: -1 isGC: false

The axis count is coming back negative.

DonLakeFlyer, 2017-01-14:
@tubeme Can you follow the instructions in #4425 to get log output? Run qgc like this from command line:
```qgroundcontrol --logging:JoystickLog```

Need to see if you are seeing the same negative axis count.

NaterGator, 2017-01-14:
@DonLakeFlyer it works fine for me on windows with the gamepads and joysticks I have available. @tubeme can you get us the JoystickLog? SDL is having trouble detecting your joystick properly... negative numbers coming from SDL indicate an error in the lib which is why you can't calibrate. 

We need to handle the error condition better (inform the user) and ideally figure out if things can be made to work with your joystick. SDL didn't change from the Dec 20th version that you said was working so it seems something more nuanced is likely going on.

nicolaerosia, 2017-01-14:
Hi, could you please take a look at my issue https://github.com/PX4/Firmware/issues/6337 ? I'm wondering if is QGroundControl related or not

tubeme, 2017-01-14:
@DonLakeFlyer I cannot try stable because it does not start. The stable is prior the #4425 fix. The master I use is compiled by @NaterGator.

Will get the log tomorrow once in the workshop.

BTW the Joystick was perfectly working in the 20dec master (3.0.2)

tubeme, 2017-01-16:
@DonLakeFlyer Here it is:

QGroundControl.exe	JoystickLog: Available joysticks
QGroundControl.exe	JoystickLog: 	 "Mad Catz F.L.Y.5 Stick" axes: -1 buttons: -1 hats: -1 isGC: false
QGroundControl.exe	JoystickLog: _loadSettings  "Mad Catz F.L.Y.5 Stick"
QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true

rjehangir, 2017-01-16:
@tubeme We're you using the same joystick when you tested with 3.0.2?

NaterGator, 2017-01-16:
Also @tubeme can you get the same log with a working version of QGC?

tubeme, 2017-01-16:
@NaterGator I will do this tomorrow when in the workshop.

@rjehangir Did you try this joystick with the 3.1.1??

tubeme, 2017-01-17:
Here it is the working system 3.0.2 master 20 dec

QGroundControl.exe	JoystickLog: Available joysticks
QGroundControl.exe	JoystickLog: 	 "Mad Catz F.L.Y.5 Stick" axes: 5 buttons: 14 hats: 1
QGroundControl.exe	JoystickLog: _loadSettings  "Mad Catz F.L.Y.5 Stick"
QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:badsettings false 0 false
QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 0 -32768 32767 0 false false
QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 1 -32768 32767 0 false false
QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 2 -32768 32767 0 false false
QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 3 -32768 32767 0 false false
QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 4 -32768 32767 0 false false
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true
QGroundControl.exe	JoystickLog: _loadSettings button:action 0 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 1 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 2 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 3 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 4 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 5 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 6 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 7 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 8 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 9 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 10 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 11 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 12 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 13 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 14 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 15 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 16 ""
QGroundControl.exe	JoystickLog: _loadSettings button:action 17 ""

rjehangir, 2017-01-17:
It looks like this happened during the switch from SDL1.2 to SDL2. The 3.0.2 version used SDL1.2 and the 3.1+ versions use SDL2. I'll look into it today and see why it's showing up with no axes or buttons.

@jaxxzer

nicolaerosia, 2017-01-17:
There seems to be a migration guide: https://wiki.libsdl.org/MigrationGuide#Joysticks
Maybe it helps. My joystick is not even detected, https://github.com/mavlink/qgroundcontrol/issues/4446

rjehangir, 2017-01-17:
In the migration guide:

> Support for the older joystick API (/dev/input/js*) for Linux has been dropped from SDL2. SDL2 only supports the newer events API (/dev/input/event*) for joysticks. These events are not normally readable for normal user accounts, so even if joysticks are plugged in you will likely have none detected. This is something that end users will have to configure for themselves.

A few people have mentioned issues and all of them, as far as I know, have been using fairly old gamepads or joysticks.

nicolaerosia, 2017-01-17:
```
These events are not normally readable for normal user accounts, so even if joysticks are plugged in you will likely have none detected. This is something that end users will have to configure for themselves.
```
I just fixed my problem by doing chmod o+rw /dev/input/event*, it works on latest QGroundControl!

tubeme, 2017-01-18:
@rjehangir @nicolaerosia You sure do chmod o+rw /dev/input/event* but what do i do in windows? As far as I understand it is a matter of some r/w wrights?

NaterGator, 2017-01-18:
@tubeme that does not seem to be likely for the Windows case. I'll have another PR in just a moment to try and get more information from you.

NaterGator, 2017-01-18:
@tubeme please get another log with this build: 
https://ci.appveyor.com/api/buildjobs/gxner2dtd7fgc7t9/artifacts/QGroundControl-installer.exe

NaterGator, 2017-01-19:
@vsobalski your report would also be valuable since you are experiencing the same bug

vsobalski, 2017-01-19:
@NaterGator Here is part of the error log with the parameter --logging:JoystickLog :
`5	1.861465	1096	QGroundControl.exe	JoystickLog: 	 libsdl error parsing joystick features: Joystick hasn't been opened yet`

Here is the full log: 
```
1	0.000000	1096	QGroundControl.exe	Settings location "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true
2	0.189831	1096	QGroundControl.exe	Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nJoystickLog.debug=true\n"
3	1.726891	1096	QGroundControl.exe	MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs"
4	1.859991	1096	QGroundControl.exe	JoystickLog: Available joysticks
5	1.861465	1096	QGroundControl.exe	JoystickLog: 	 libsdl error parsing joystick features: Joystick hasn't been opened yet
6	1.861561	1096	QGroundControl.exe	JoystickLog: 	 "vJoy Device" axes: -1 buttons: -1 hats: -1 isGC: false
7	1.861951	1096	QGroundControl.exe	JoystickLog: _loadSettings  "vJoy Device"
8	1.862038	1096	QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
9	1.862113	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
10	1.862184	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
11	1.862246	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
12	1.862305	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true
13	1.964892	1096	QGroundControl.exe	JoystickLog: 	 libsdl error parsing joystick features: Joystick hasn't been opened yet
14	1.964973	1096	QGroundControl.exe	JoystickLog: 	 "BU0836A Interface" axes: -1 buttons: -1 hats: -1 isGC: false
15	1.965324	1096	QGroundControl.exe	JoystickLog: _loadSettings  "BU0836A Interface"
16	1.965409	1096	QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
17	1.965516	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
18	1.965580	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
19	1.965645	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
20	1.965708	1096	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true
21	2.684527	1096	QGroundControl.exe	Map Cache in: "C:/Users/aosl/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db"
22	4.693413	1096	QGroundControl.exe	Orphaned log file count 1
23	4.693468	1096	QGroundControl.exe	Orphaned log file "C:/Users/aosl/AppData/Local/Temp/FlightData453591.mavlink"
24	4.693556	1096	QGroundControl.exe	QGCMessageBox (unit testing) "Found unsaved Flight Data" "This can happen if QGroundControl crashes during Flight Data collection. If you want to save the unsaved Flight Data, select the file you want to save it to. If you do not want to keep the Flight Data, select 'Cancel' on the next dialog."
25	4.720075	1096	QGroundControl.exe	LinkManager mavlinkStatus:channel:flags 0x15d10fc 1 2
26	8.444881	1096	QGroundControl.exe	shell\comdlg32\fileopensave.cpp(9476)\COMDLG32.dll!77C807D0: (caller: 77C73458) ReturnHr[PreRelease](1) tid(23a8) 80070490 Element not found.
27	8.444881	1096	QGroundControl.exe	    CallContext:[\PickerModalLoop\InitDialog\FileDialogInitEnterpriseData] 
28	11.106664	1096	QGroundControl.exe	LinkManager mavlinkStatus:channel:flags 0x15d1118 2 2
29	11.200391	1096	QGroundControl.exe	Temp log "C:/Users/aosl/AppData/Local/Temp/FlightData640159.mavlink"
30	14.216309	1096	QGroundControl.exe	Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 2
31	17.202350	1096	QGroundControl.exe	Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 3
32	20.207833	1096	QGroundControl.exe	Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 4
```

Let me know if you need me to run another test ( with different parameters)

NaterGator, 2017-01-19:
Thanks @vsobalski, I think this is related to the absence of a HWND context for DirectInput API calls that libsdl is likely making. I have another build for you to test if you don't mind: https://ci.appveyor.com/api/buildjobs/vsb78t4qqcqdlqr9/artifacts/QGroundControl-installer.exe 

If my suspicion is right I should have a fix ready in just a moment.

tubeme, 2017-01-19:
The same for me:

QGroundControl.exe	JoystickLog: Available joysticks
QGroundControl.exe	JoystickLog: 	 libsdl error parsing joystick features: Joystick hasn't been opened yet
QGroundControl.exe	JoystickLog: 	 "Mad Catz F.L.Y.5 Stick" axes: -1 buttons: -1 hats: -1 isGC: false
QGroundControl.exe	JoystickLog: _loadSettings  "Mad Catz F.L.Y.5 Stick"
QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true

NaterGator, 2017-01-19:
@tubeme did you try the installer from https://github.com/mavlink/qgroundcontrol/issues/4442#issuecomment-273842880

vsobalski, 2017-01-19:
Perfect, I am glad to hear my feedback is helpful. 
Here is part of the log with newer build:
```
5	1.830901	2656	QGroundControl.exe	JoystickLog: 	 libsdl failed opening joystick vJoy Device error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0xx
```

Here is the full log trace:

```
1	0.000000	2656	QGroundControl.exe	Settings location "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true
2	0.242876	2656	QGroundControl.exe	Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nJoystickLog.debug=true\n"
3	1.764622	2656	QGroundControl.exe	MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs"
4	1.828790	2656	QGroundControl.exe	JoystickLog: Available joysticks
5	1.830901	2656	QGroundControl.exe	JoystickLog: 	 libsdl failed opening joystick vJoy Device error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0xx
6	1.831732	2656	QGroundControl.exe	JoystickLog: 	 libsdl failed opening joystick BU0836A Interface error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0xx
7	2.420278	2656	QGroundControl.exe	Map Cache in: "C:/Users/aosl/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db"
8	4.108493	2656	QGroundControl.exe	Orphaned log file count 1
9	4.108606	2656	QGroundControl.exe	Orphaned log file "C:/Users/aosl/AppData/Local/Temp/FlightData640159.mavlink"
10	4.108740	2656	QGroundControl.exe	QGCMessageBox (unit testing) "Found unsaved Flight Data" "This can happen if QGroundControl crashes during Flight Data collection. If you want to save the unsaved Flight Data, select the file you want to save it to. If you do not want to keep the Flight Data, select 'Cancel' on the next dialog."
11	4.219568	2656	QGroundControl.exe	LinkManager mavlinkStatus:channel:flags 0x12110fc 1 2
12	9.842532	2656	QGroundControl.exe	LinkManager mavlinkStatus:channel:flags 0x1211118 2 2
13	9.874044	2656	QGroundControl.exe	Temp log "C:/Users/aosl/AppData/Local/Temp/FlightData069227.mavlink"
14	11.884721	2656	QGroundControl.exe	shell\comdlg32\fileopensave.cpp(9476)\COMDLG32.dll!77C807D0: (caller: 77C73458) ReturnHr[PreRelease](1) tid(8d0) 80070490 Element not found.
15	11.884721	2656	QGroundControl.exe	    CallContext:[\PickerModalLoop\InitDialog\FileDialogInitEnterpriseData] 
16	12.880977	2656	QGroundControl.exe	Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 2
17	15.882086	2656	QGroundControl.exe	Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 3
18	18.884481	2656	QGroundControl.exe	Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 4

```

NaterGator, 2017-01-19:
Thanks again @vsobalski, exactly what I expected. Can you let me know if https://ci.appveyor.com/api/buildjobs/06yrg05ynpto2j0s/artifacts/QGroundControl-installer.exe works?

NaterGator, 2017-01-19:
Okay... @tubeme @vsobalski try this one and let me know how it goes: https://ci.appveyor.com/api/buildjobs/d7nut8h9wpgynt11/artifacts/QGroundControl-installer.exe

tubeme, 2017-01-19:
I will try the last one tomorrow.

vsobalski, 2017-01-20:
The last version seems to detect both the virtual joystick and the real joystick. 
Here is part of the log output:

```
6	7.631393	8360	QGroundControl.exe	JoystickLog: Available joysticks
7	7.690119	8360	QGroundControl.exe	JoystickLog: 	 "vJoy Device" axes: 8 buttons: 8 hats: 0 isGC: false
8	7.690632	8360	QGroundControl.exe	JoystickLog: _loadSettings  "vJoy Device"
9	7.690721	8360	QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
10	7.690815	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 0 -32768 32767 0 false 0 false
```

The GroundStation doesn't open.

here is the full log output : 

```
1	0.000000	8360	QGroundControl.exe	Settings location "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true
2	0.013478	8360	QGroundControl.exe	Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nJoystickLog.debug=true\n"
3	4.658179	8360	QGroundControl.exe	MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs"
4	4.990234	8360	QGroundControl.exe	Map Cache in: "C:/Users/aosl/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db"
5	7.517681	8360	QGroundControl.exe	Orphaned log file count 0
6	7.631393	8360	QGroundControl.exe	JoystickLog: Available joysticks
7	7.690119	8360	QGroundControl.exe	JoystickLog: 	 "vJoy Device" axes: 8 buttons: 8 hats: 0 isGC: false
8	7.690632	8360	QGroundControl.exe	JoystickLog: _loadSettings  "vJoy Device"
9	7.690721	8360	QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
10	7.690815	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 0 -32768 32767 0 false 0 false
11	7.690917	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 1 -32768 32767 0 false 0 false
12	7.691002	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 2 -32768 32767 0 false 0 false
13	7.691086	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 3 -32768 32767 0 false 0 false
14	7.691215	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 4 -32768 32767 0 false 0 false
15	7.691279	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 5 -32768 32767 0 false 0 false
16	7.691371	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 6 -32768 32767 0 false 0 false
17	7.691457	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 7 -32768 32767 0 false 0 false
18	7.691536	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
19	7.691613	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
20	7.691744	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
21	7.691825	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true
22	7.691962	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 0 ""
23	7.692046	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 1 ""
24	7.692135	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 2 ""
25	7.692245	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 3 ""
26	7.692314	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 4 ""
27	7.692376	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 5 ""
28	7.692435	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 6 ""
29	7.692495	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 7 ""
30	7.753808	8360	QGroundControl.exe	JoystickLog: 	 "BU0836A Interface" axes: 4 buttons: 32 hats: 1 isGC: false
31	7.754301	8360	QGroundControl.exe	JoystickLog: _loadSettings  "BU0836A Interface"
32	7.754399	8360	QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
33	7.754492	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 0 -32768 32767 0 false 0 false
34	7.754580	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 1 -32768 32767 0 false 0 false
35	7.754666	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 2 -32768 32767 0 false 0 false
36	7.754751	8360	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 3 -32768 32767 0 false 0 false
37	7.754828	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
38	7.754910	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
39	7.754997	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
40	7.755067	8360	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true
41	7.755143	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 0 ""
42	7.755220	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 1 ""
43	7.755272	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 2 ""
[... cut ...]
72	7.757627	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 31 ""
73	7.757714	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 32 ""
74	7.757800	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 33 ""
75	7.757907	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 34 ""
76	7.758003	8360	QGroundControl.exe	JoystickLog: _loadSettings button:action 35 ""
77	7.909161	8360	QGroundControl.exe	LinkManager mavlinkStatus:channel:flags 0x19f10fc 1 2

```

NaterGator, 2017-01-20:
@vsobalski I'm struggling with the failure to load issue. If you go back to the master build does it still fail to load?

vsobalski, 2017-01-20:
The Original version ( from http://qgroundcontrol.com/) dosen't start, but the version from issue #4425 works on my computer. 

Would a log output with specific parameters help debugging it?

NaterGator, 2017-01-20:
Are you running with --logging:full?

vsobalski, 2017-01-20:
I didn't at first, but I just restarted with -- logging:full.

I enclosed it in a file, as there is a lot of output

[QGroundControl-LogViewFull1.txt](https://github.com/mavlink/qgroundcontrol/files/720170/QGroundControl-LogViewFull1.txt)

NaterGator, 2017-01-20:
Thanks. That didn't appear to recognize your joystick... is that the older version that is starting up successfully? Can you try the newest build in this issue that wasn't opening but was detecting your joystick?

tubeme, 2017-01-20:
Works for me perfect. I was able to calibrate the Joystick.

11	26.310559	5684	QGroundControl.exe	JoystickLog: Available joysticks
12	26.315473	5684	QGroundControl.exe	JoystickLog: 	 "Mad Catz F.L.Y.5 Stick" axes: 5 buttons: 14 hats: 1 isGC: false
13	26.315999	5684	QGroundControl.exe	JoystickLog: _loadSettings  "Mad Catz F.L.Y.5 Stick"
14	26.316113	5684	QGroundControl.exe	JoystickLog: _loadSettings calibrated:throttlemode:exponential:deadband:badsettings false 0 false false false
15	26.316221	5684	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 0 -32768 32767 0 false 0 false
16	26.316319	5684	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 1 -32768 32767 0 false 0 false
17	26.316415	5684	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 2 -32768 32767 0 false 0 false
18	26.316500	5684	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 3 -32768 32767 0 false 0 false
19	26.316595	5684	QGroundControl.exe	JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 4 -32768 32767 0 false 0 false
20	26.316690	5684	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 0 -1 true
21	26.316762	5684	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 1 -1 true
22	26.316835	5684	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 2 -1 true
23	26.316917	5684	QGroundControl.exe	JoystickLog: _loadSettings function:axis:badsettings 3 -1 true
24	26.317025	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 0 ""
25	26.317126	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 1 ""
26	26.317239	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 2 ""
27	26.317339	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 3 ""
28	26.317428	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 4 ""
29	26.317529	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 5 ""
30	26.317630	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 6 ""
31	26.317705	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 7 ""
32	26.317788	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 8 ""
33	26.317846	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 9 ""
34	26.317909	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 10 ""
35	26.317970	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 11 ""
36	26.318032	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 12 ""
37	26.318094	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 13 ""
38	26.318156	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 14 ""
39	26.318218	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 15 ""
40	26.318279	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 16 ""
41	26.318341	5684	QGroundControl.exe	JoystickLog: _loadSettings button:action 17 ""

NaterGator, 2017-01-20:
@tubeme that is good to hear!
@vsobalski I think the issue you are having is not related to this original issue. Can you open a new issue?

vsobalski, 2017-01-20:
@NaterGator No problems.

NaterGator, 2017-01-20:
Thanks everyone. @DonLakeFlyer  #4452 is ready to merge.

closed 2017-01-20 by NaterGator

johnlongland, 2021-01-26:
Good day all
I have been reading this discussion #4442 about the joystick that cannot be calibrated. I see it was back in 2017.
I have just purchased a new remote control system from CUAV that uses QGC. I am assuming it is runing on Android and I am having the same issue. If I read what you guys have been doing it leaves me pretty hopeless because I do not have access onto the operating system. The manufactorers installed QGC and I am not able to load any newer versions of QGC.  I see the discussion i smarked closed so I am not even sure if my request is going to be logged. Many thanks