mavlink/qgroundcontrol #4445: Integrate crash reporting tools

NaterGator, 2017-01-16:
At least on non-mobile applications we would be well served to handle crashes more gracefully. As mentioned [here](https://github.com/mavlink/qgroundcontrol/pull/2680#issuecomment-173740648) a decent choice would be the Chromium project's [breakpad](https://chromium.googlesource.com/breakpad/breakpad/) library. 

We can start small and just generate crash dumps when an unhandled exception occurs (users would have to manually upload them). Later down the line it would be nice to have a built in console / tool to facilitate users submitting stored crash reports.

NaterGator, 2017-08-11:
@DonLakeFlyer what do you think about integration with https://drdump.com ? We can upload the release build symbols to their service and collect crash dumps from Windows there.

DonLakeFlyer, 2017-08-14:
That sound like a great idea. Do we need to start flipping some bits in the installer to get better crash reports?

jaxxzer, 2017-10-17:
@NaterGator have you done any work on this? I'm interested in helping get this together.

NaterGator, 2017-10-17:
I looked into this and we would have to package binary blobs that we can't audit from a software company in Russia.  Didn't / doesn't feel right. I'd like to find an open-source and auditable solution to this problem.

DonLakeFlyer, 2018-04-06:
Breakpad support is working it's way into 3.4 release hopefully

patrickelectric, 2018-11-15:
Hey guys, any update on it ? 
probably waiting for 5.12 to have the debug symbols ?
Any motive to use breakpad and not other project ?

patrickelectric, 2018-11-15:
I'm adding a crash report for one application, you can see the output window here:
[image]

After doing the deploy and tests for windows , linux and mac. I'll add it in QGC.

DonLakeFlyer, 2018-11-15:
Would be fantastic to get this all working cross-platform. Huge win.

jaxxzer, 2019-02-05:
We are doing some work with [KCrash](https://api.kde.org/frameworks/kcrash/html/index.html
) and [drkonqi](https://github.com/KDE/drkonqi
)

We do intend to attempt some integration with QGC. Will let you all know how that goes.

DonLakeFlyer, 2019-02-05:
The listed support for KCrash is "FreeBSD, Linux, Windows". Does that include OSX and Android?

patrickelectric, 2019-02-05:
The only unsupported OS is Android for now, I did some work and tests with this patch: https://phabricator.kde.org/D4929, and we got it merged last year for OSX.
Android will be supported soon since KDE is strongly working to improve plasma mobile support and kirigami applications.

patrickelectric, 2019-02-05:
Well, if that is possible, I would like very much to take a look on it and to help with the others OS integration in QGC :)

DonLakeFlyer, 2019-02-05:
Ok, let me ask for permission.

jaxxzer, 2019-02-05:
> If you want I can ask permission to bring that into the main repo.

Please do. This is exactly why I bring up our own work here: In order to align efforts.

jaxxzer, 2019-07-18:
We are continuing to work with integrating kcrash with qgc. @tcanabrava can provide more details about that. @DonLakeFlyer please let us know if you can share the breakpad + android project. In any event, we can use kcrash for the other platforms. Problems on Windows machines have been my main concern. We've had good luck everywhere else.

DonLakeFlyer, 2019-07-18:
> please let us know if you can share the breakpad + android project

Sorry. I never got permission for that.

What exactly does kcrash provide as far as features? FYI: The Windows builds already have crash dump capture turned on for them by the installer. Which you can then load into VS.

jaxxzer, 2019-07-18:
> What exactly does kcrash provide as far as features? 

@tcanabrava can help to explain it better, but as I understand it, kcrash provides the facility to collect the stack trace, and drkonqi is an optional program that can be installed independently of qgc and can provide the 'program crashed' ui that presents the stack trace to the user and gives them the option to report it.

> FYI: The Windows builds already have crash dump capture turned on for them by the installer. Which you can then load into VS.

Thank you for the heads up. I have used this process before. I found it exceedingly difficult and time consuming to get right with the .pdbs from S3. The crash dump files are also huge (> hundreds of MB). These crash dumps provide a more introspective option for debugging, but the stack trace will provide a quick and easy debugging option and facilitate identifying patterns.

DonLakeFlyer, 2019-07-18:
> The crash dump files are also huge (> hundreds of MB).

Also just FYI: I changed them to be minidumps a while back.

> drkonqi is an optional program that can be installed independently of qgc and can provide the 'program crashed' ui that presents the stack trace to the user and gives them the option to report it.

Hmm. Installing another app on the machine to make this all work is a bit concering.

jaxxzer, 2019-07-18:
AFAIK:
kcrash is the only dependency we need to capture the stack trace. I think we could simply write that report to a file and tell the user where it is (@tcanabrava can clarify).
drkonqi is an optional dependency that presents a UI with the kcrash report. drkonqi could be bundled with QGC, or installed separately to reduce bloat.

tcanabrava, 2019-07-18:
@DonLakeFlyer KCrash + DrKonqi are a combination of thigns.
KCrash is a crash retrieval feature that supports generation of backtraces in mac / linux / windows / iOS (and android is comming), it's fully opensource and used in almost all softwares from the KDE community. for your app to support KCrash, you basically call two functions after the creation of QApplication (and of course, link to the library)

DrKonqi is the front-end that displays to the user the backtrace and asks for permission to create a bug report. I started fixing bugs in it this month so it's easier integrated in non-kde projects. (this is not finished, but I hope to finish soon)

The usability goes like this:
QGC Crashes
DrKonqi displays the backtrace
User clicks in "Create Bugreport" 
The backtrace + User comments are submited to the bugracker.

What I still need to do in DrKonqi for a correct integration in QGC:
- Support Github Tickets (currently it only supports bugzilla)
- Support Per-Project configuration (Currently it will send to bugs.kde.org by default)

jaxxzer, 2019-07-23:
> Support Github Tickets (currently it only supports bugzilla)

@tcanabrava Is this optional? Could an email address be used instead?

tcanabrava, 2019-09-17:
not optional, we are working in backends for it.

jaxxzer, 2019-09-17:
@DonLakeFlyer @tcanabrava 

I recommend we forgo the drkonqi UI + automatic reporting tools right now (they need require further development).

We can use kcrash only (the user only needs to install qgc), and when a crash occurs, write a stack trace to a text file (perhaps under Documents/QGC/crashlog). When a user reports a crash, we can request them to retrieve this file on a case by case basis.

DonLakeFlyer, 2019-09-20:
> I recommend we forgo the drkonqi UI + automatic reporting tools right now

Sounds fine.

I'm interesting in seeing what sort of data you get back from this though. @patrickelectric Gave an example but it is fully Qt based. For example on Windows without a PDB you aren't going to get function names for QGC code will you?

tcanabrava, 2019-09-21:
No, a pdb is needed. But the pdb could be downloaded via the interface when
the user explicitly clicked a button for it

DonLakeFlyer, 2019-09-22:
That really makes me wonder about the usefulness of this then. For example is the effort more well spent to get QGC Windows into the store. Which will both gets us nice server based crash reporting as well as store support. Better upgrade support and so forth. Might be just as much work to do that. Kill two birds with one stone.

tcanabrava, 2019-09-23:
I can't say, Don, as I don't use windows for almost 20 years. Does the
store support on windows also helps the Mac and Linux codebases?

DonLakeFlyer, 2019-09-23:
> Does the store support on windows also helps the Mac and Linux codebases?

No it doesn't. But the fact is that Windows is the largest use case and also tends to be the OS which is the quirkiest. Hence information on crashes caught there will benefit all OS. It is very rare for there to be a crash which is OSX or Linux specific. So if you can get the prevalent desktop build reporting that will cover most everything.

jaxxzer, 2019-09-23:
Solid point about the pdb and so on @DonLakeFlyer. We will look into the store options you mentioned and bring them into consideration.

DonLakeFlyer, 2019-09-23:
The thing is we already have iOS and Android covered for crash reporting on mobile since they are in the store. We are just missing a desktop version.