mavlink/qgroundcontrol #4446: Joystick not working on master

nicolaerosia, 2017-01-16:
Hello,

I have joystick working on v3.0.2 but it is not working on v3.1.1/v3.1.2 or daily build.
OS: Linux, Fedora 25 x86_64.
Is the project bisectable? How can I help?

Thanks,
Nicolae

NaterGator, 2017-01-17:
@nicolaerosia can you see if your issue is related to https://github.com/mavlink/qgroundcontrol/issues/4442

nicolaerosia, 2017-01-17:
Tried daily build (17-01-2017) with Joystick logging on:
```
[user@mhome qgroundcontrol-17-01-2017]$ ./qgroundcontrol-start.sh --logging:JoystickLog
Settings location "/home/user/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nJoystickLog.debug=true\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/user/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
JoystickLog: Available joysticks
VideoReceiver::start() failed. Error with gst_element_factory_make('avdec_h264')
VideoReceiver::start() failed
Map Cache in: "/home/user/.cache/QGCMapCache300" / "qgcMapCache.db"
Orphaned log file count 0
LinkManager mavlinkStatus:channel:flags 0xebdfe8 1 2
Temp log "/tmp/FlightData737780.mavlink"
```

NaterGator, 2017-01-17:
Can you provide the output of lsusb?

nicolaerosia, 2017-01-17:
Sure
```
Bus 001 Device 003: ID 0603:1a13 Novatek Microelectronics Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0603 Novatek Microelectronics Corp.
  idProduct          0x1a13 
  bcdDevice            3.01
  iManufacturer           0 
  iProduct               34 ART TECH GAME.  
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              296mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      40
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              20
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

nicolaerosia, 2017-01-17:
I fixed my problem by doing `chmod o+rw /dev/input/event*`, I'll create a udev rule to fix my permission problems. This is explained here: https://wiki.libsdl.org/MigrationGuide#Joysticks

NaterGator, 2017-01-17:
Good to hear!

closed 2017-01-17 by NaterGator