mavlink/qgroundcontrol #4447: QGroundControl won't start after being run as root

nicolaerosia, 2017-01-16:
Hello,

It seems that QGroundControl won't start as a regular user after it was run as root because it cannot access the files already created by root in /tmp.
The problem is that QGroundControl just exits, without any message.

```
[user@mhome qgroundcontrol-3.1.1]$ ls -l /tmp/qipc_s*
-rw-r-----. 1 root root 0 Jan 16 22:31 /tmp/qipc_sharedmemory_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd
-rw-r-----. 1 root root 0 Jan 16 22:31 /tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd
-rw-r-----. 1 root root 0 Jan 16 22:31 /tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed
```

strace output:
```
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a648b, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a648b, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a648b, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a648b, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_adccebdaceeeebae17280624d1270313ae1b5ea6ccc6b351d44aacbd", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a648b, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a648b, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] open("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0640) = -1 EEXIST (File exists)
[pid  7849] stat("/tmp/qipc_systemsem_bcbcfdefbcbebf9df1f53a47c9e16d877a0e930daaf24287da47ed", {st_mode=S_IFREG|0640, st_size=0, ...}) = 0
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|IPC_EXCL|0600) = -1 EEXIST (File exists)
[pid  7849] semget(0x512a6488, 1, IPC_CREAT|0600) = -1 EACCES (Permission denied)
[pid  7849] exit_group(0)               = ?
```

DonLakeFlyer, 2017-01-17:
I doubt this is something we are going to fix inside of QGC. Running apps as root on linux leads to all sorts of havoc. You'll have to clean up the permissions, delete installations and re-install to fix.

closed 2017-01-17 by DonLakeFlyer

nicolaerosia, 2017-01-17:
I've run it to see if it detects Joystick but the application warned me about running as root and exited without cleaning up the files in tmp.