mavlink/qgroundcontrol #4449: PlayUAVOSD: Persistent problem.

tubeme, 2017-01-17:
Hello,

We had flown our PlayUAVOSD with older versions of QGC and PX4. We connect the telemetry to the TELEM 2 port and the system param sys_companion to OSD 57600
But when we connected to the QGC it gives continues errors:

[image]

[image]

The Telemetry on the OSD is working perfect. The horizon the data is on the video. So the PlayUAVOSD is receiving the data.

May be it is on the QGC side.

Here is the log I get with DebugView++ making a --logging:full while receiving this error.

https://drive.google.com/file/d/0B5qO0rf9mnFPaXNPSlk4R1EtTUE/view?usp=sharing

tubeme, 2017-01-19:
I saw this from Dec 2016.

http://en.playuav.com/question/45

NaterGator, 2017-01-19:
Fixed in PX4 firmware: https://github.com/PX4/Firmware/commit/1c42cea28e6e09c75ed2c8ab5c012ee4465f9ffb

closed 2017-01-19 by NaterGator

tubeme, 2017-01-19:
Thanks @NaterGator Maybe it has something to do with the OSD functionality. The OSD reads the mission from the px4 and then has a map for the mission. So probably they did not follow some Mavlink rules or sth.?

What can I write to them in order to fix the problem? They can make correction to their code as well.