mavlink/qgroundcontrol #4451: Coping with RC_Channel split

tridge, 2017-01-18:
Hi Don,
This issue is to create a place to discuss the implications of the RC_Channel library split in ArduPilot master.
The changes are:

-  RCn_REV becomes RCn_REVERSED, and becomes a 0 or 1 boolean
-  RCn_MIN/MAX/TRIM only affect inputs. This was always the case on copters, but is now true on planes/rovers too
-  for planes/rovers the output trims are in SERVOn_TRIM
- the old RCn_FUNCTION parameters are now SERVOn_FUNCTION (as they affect output, not input)

DonLakeFlyer, 2017-01-19:
Ok, read the comments on the RCn_REV changed. I didn't initially pick on the fact that it was more than a name change.

Are RCn_FUNCTION name changes only? 

Currently QGC sets MIN/MAX as the min/max PWM values it sees during it calibration process. I assume this is still ok.

For _TRIM values QGC reads the PWM values for the centered sticks and sets the _TRIM params to those values. Is that still appropriate?

tridge, 2017-01-22:
> Ok, read the comments on the RCn_REV changed. I didn't initially pick on the fact that it was more than a name change.

yes, I changed the name to give a way to distinguish the new semantics from the old. I thought that GCS implementation could check for the existence of SERVO1_REVERSED to know if the knew behaviour is needed.

> Are RCn_FUNCTION name changes only?

yes, although you will notice that all channels have a FUNCTION now, even first 4 channels. Previously only channels 5 and above had a FUNCTION. So you can now assign any function to any output channel.

> Currently QGC sets MIN/MAX as the min/max PWM values it sees during it calibration process. I assume this is still ok.
> For _TRIM values QGC reads the PWM values for the centered sticks and sets the _TRIM params to those values. Is that still appropriate?

that sounds right, thanks!

DonLakeFlyer, 2017-01-23:
Ok, this is all fairly easily doable from q QGC standpoint then.

lucasdemarchi, 2017-02-14:
> RCn_MIN/MAX/TRIM only affect inputs. This was always the case on copters, but is now true on planes/rovers too

AFAICS RC3_MIN/MAX still applies to output if MOT_PWM_MIN/MAX are not set or set to 0.

DonLakeFlyer, 2017-02-14:
@tridge Can you tell me the firmware version numbers for plane/rover for which this new param naming starts. For example for Copter it's 3.5. It need it to fill out my parameter remapping arrays correctly.

DonLakeFlyer, 2017-02-15:
@tridge I figured out the versions. I can ge them from the dev firmware download naming.

closed 2017-02-15 by DonLakeFlyer