mavlink/qgroundcontrol #4457: Windows: On Start QGC doesn't load

vsobalski, 2017-01-20:
Good Afternoon,

Following the issues #4425 and #4442 I am still experiencing issues with QGC.

The issue  #4425 resolved the start problem, and I was able to start QGC, but the joystick wasn't recognize.
The issue  #4442 resolved the joystick problem, but I was unabled to start QGC.

Enclosed is a log issued from the last version of issue #4442 using DebugView with the parameter --logging:full
[QGroundControl-LogViewFull2.txt](https://github.com/mavlink/qgroundcontrol/files/720219/QGroundControl-LogViewFull2.txt)

Let me know if you need any additional information.

Thanks a lot for your help,

Vincent

NaterGator, 2017-01-20:
Can you try clearing your map cache?

vsobalski, 2017-01-23:
Clearing the cache work, thanks a lot for your help. This issue is then resolved.

I have one more question, not related to the issue, I hope it's OK if I ask it here. I have a custom AUV. When I connect it to QGC, the joystick doesn't appear. But if I use my PixHawk, it appears right away. Is there a signal that QGC is looking for to enable the joystick tab?

NaterGator, 2017-01-23:
Good! The joystick button in the setup view is set visible by the following conditions:
https://github.com/mavlink/qgroundcontrol/blob/f1b82fb9d2fc8ce20c5656b54ba500db9f7b1af4/src/VehicleSetup/SetupView.qml#L308

Since you have a nonzero number of joysticks, _fullParameterVehicleAvailable is the likely culprit. That variable is set here:
https://github.com/mavlink/qgroundcontrol/blob/f1b82fb9d2fc8ce20c5656b54ba500db9f7b1af4/src/VehicleSetup/SetupView.qml#L42

So most likely for whatever reason QGC doesn't think it is getting all the available parameters from your custom AUV. Hope that helps!

closed 2017-01-23 by NaterGator

vsobalski, 2017-01-23:
Thanks a lot for your feedback. I had a problem in my parameters, which was the reason why the joystick wasn't showing up. It's working great now.
Have a nice day,
Vincent