mavlink/qgroundcontrol #4461: Switching to ACRO mode with disabled THR failsafe couses crash

TanoStaf, 2017-01-23:
If necessary to fly without radio, just with laptop, by mission you should turn of Throttle fail-safe. To start mission you need to switch to AUTO mode in the air. There are no confirm slider (or whole mode lock on flying without radio) on dummy switching to ACRO mode, which definitely drops down your copter.

DonLakeFlyer, 2017-01-23:
Not sure I understand what you want here. Turn off throttle-fail safe? What exactly do you mean by that. What exact setting/param where?

TanoStaf, 2017-01-24:
In Ardupilot you have two failsafe options - on telemetry data connection loss and\or on radio signal loss (thru monitoring throttle PWM signal on ch3 not going lower some treshold level (900 ms by default)). If you need to control copter just by laptop - you need to disable radio loss failsafe becouse you will got 0 PWM level on ch3 all time. After this only one thing you need - it just controll copter by connecten gamepad, or go by points, or run mission flight. If you did that and changing to any fly mode with "root" control like ACRO|SPORT ets it the air - copter definitely falls down course it have 0 throttle input signal sue to unconnected radio receiver.

I can't imagine any case where it necessary to do this, but it still possible without any warning or safety sliders ets.

TanoStaf, 2017-01-24:
"Turn off throttle-fail safe?" No. My suggestions are:
1) Make combobox list of flight modes editable to remove unnecessary items for pilot. 
2) Add confirmation slider on flight mode change to ARCO or SPORT.
3) Forbid changes to "root" modes (which definitely drops copter) when radio signal not exists and failsafe on throttle if off.

DonLakeFlyer, 2017-01-31:
It still seems like a strange setup. As far as I know ArduPilot doesn't support flying without some sort of controller (does it?). Either a real RC transmitter or a simulated one from a joystick/gamepad. Also at this point QGC doesn't support flying ArduPilot with a joystick.

If there is a way to fly ArduPilot with no rc transmitter or joystick then I'm not sure how QGC would know if you are flying this way. It would be much more of the firmwares job to disallow a flight mode change which could result in the vehicle going haywire than QGC preventing that. There are too many variables involved for QGC to make those sort of decisions. The firmware should fail the flight mode change when asked to. Just like it does for other commands like arming when it isn't ready and so forth.

If this is really something you need I'd as the ArduPilot firmware folks to support #3 on the firmware side by forbidding the mode change. They have full information to make those sorts of decisions where QGC does not.

TanoStaf, 2017-01-31:
1) Yes, it supports flight just by mission commands with no  joystic simulation.
2) QCG shouldn't know about this (but can - by monitoring throttle signal level that can't be less than 900 ms). It's more than enough to add sliders on "unsafe" flight mode and add some warning when throttle is 0 and you going no change plan and drop your MAV down:)  If we want to stop motors in the air - we already have kill switch)

Sure it's not obvious about side of this sort of prevention. For my opinion it's looks like root mode in Linux. Linux kernel provides you anything you want to do, but just after (password) confirmation. It's not the kernels privilege to block your unsafe actions, maybe you know what you are doing. But Userspace asks a confirmation. Same with APM - It should do everything you ask, but if you want to do somethings unsafe - just confirm.

The easiest way to solve this problem is to make able to edit and hide items of flight modes dropdown list. Maybe in "flight modes" section of Settings? As I know, most of user-pilots using about 2-4 modes depending of setup. Another can be not just useless, but also dangerous, as we see.

DonLakeFlyer, 2017-01-31:
Ok, understand more now. This should be a firmware side fix to prevent a bad flight mode change not a groundstation feature. There are too many variables for a groundstation to understand to correctly implement this. You can also control a vehicle using an offboard controller in which case it is likely ok to change to any flight mode even if there is no rc signal since the offboard controller is controlling the vehicle attitude. There are likely other cases where it's perfectly fine to switch flight modes when there is no rc. For example the mavlink MANUAL_CONTROL messages allow you to control a vehicle without an RC connected. Only the firmware can ultimately be in control of failing a change in flight mode. The ground station doesn't know enough about the firmware implementation to be able to do it correctly.

closed 2017-01-31 by DonLakeFlyer