mavlink/qgroundcontrol #4464: Qgroundcontrol 3.1.2 will not replay telemetry log files

Kjkinney, 2017-01-24:
https://drive.google.com/file/d/0ByGQI8gP4hzqYjZQWGg2VHZ1aHc/view?usp=sharing

here is a link to the file I am trying to play. For some reason it shows a very large amount of time. I have tested this file on older versions of qgroundcontrol and it works just fine.

[image]

Kynos, 2017-03-14:
Same here with 3.1.3, .mavlink file attached. 

Just a wild guess: might be caused by a non-valid GPS time (this ia a short indoor flight).

[disarmed.zip](https://github.com/mavlink/qgroundcontrol/files/842574/disarmed.zip)

DonLakeFlyer, 2017-05-18:
This is working now

closed 2017-05-18 by DonLakeFlyer