mavlink/qgroundcontrol #4469: Errors inserting and deleting waypoints

Antiheavy, 2017-01-24:
I'm fairly new to QGC, I hope this is the right place to post this:

I am seeing some errors when inserting and deleting waypoints.  I tested this in both QGC 3.0.2 and QGC v3.1.2  on two different in Windows 7 64-bit machines.  Also this happens both when connected to a PX4/Pixracer and when not connected to any autopilot.  I'm not sure if this is one or two separate issues:

1) sometimes (not always) when I insert a waypoint it comes in at altitude = 0.  In the image below I "inserted" waypoint 5 and it showed up at (near?) the HOME location at altitude=0; 
[image]
Having a waypoint automatically set (or sometimes set) to altitude = 0 seems undesirable with the important exception of a LAND waypoint.

2) When I delete an inserted waypoint (whether it came it at alt = 0 or alt = 50m) some graphic errors show up with the mission.  For example the picture below I deleted the previously inserted waypoint 5, now there is no "5" marker on the map, but the mission lines still go to where new "5" should be and there is a blank circle where the old previously inserted "5" was:
[image]

Further deleting of waypoints can produce other graphical errors:
[image]

And sometimes while inserting and deleting waypoints GGC crashes:
[image]
Does QGC create logs that might help debugging?

Thank you for your consideration!

Antiheavy, 2017-01-25:
Update:  I had a chance to test this on a Mac (Sierra OS).  Fresh installed Qgc v3.1.2.  The same waypoint inserting and deleting bugs occur.

DonLakeFlyer, 2017-01-31:
The delete problems are a known problem for which there is an issue entered. Unfortunately it is a bug in the Qt library we are using. Still trying to find a workaround. 

Your #1, 0 altitude problem I have never heard of before. If you could figure out a set of steps to reproduce that would be awesome. Then I can fix.

Antiheavy, 2017-01-31:
Thanks for your reply.  Ooops, I guess I tried to see if the waypoint insert issue was already reported, but I must not have searched very well.  Bummer about the Qt bug.

I figured out my 0 altitude problem when inserting:  1)  I plan several waypoints, 2) i change the last waypoint to a Land waypoint, which comes in at a non 0 altitude, 3) I (hopefully) remember to set it to 0 altitude, 4) I then insert a waypoint somewhere else and it comes it with 0 altitude.

I believe this has something to do with how Qgc is grabbing the most recently used altitude.  I see in some other forum posts that the way Qgc handles default waypoint altitude has changed several times.  It also looks like folks are working on setting the Land point to 0 by default in a future release.

Somewhat off topic:
if I could make a humble request I'd ask for a feature that allowed the user to set a default waypoint altitude (maybe in the Application Settings -> General screen).  Possibly "use previous waypoint altitude" functionality could be optionally disabled/enabled with a checkbox in the same Application Settings screen?  Land waypoint would need to be handled as a special case that avoids my inserting problem above, but still defaults Land to 0.  Anyway, I understand you need to balance the desires of a large community.

Cheers.

DonLakeFlyer, 2017-03-19:
Insert/delet problems fixed by move to Qt 5.7

closed 2017-03-19 by DonLakeFlyer