mavlink/qgroundcontrol #4474: Help With Gimble Control

stefanoghazzali, 2017-01-26:
I have a PixHawk Board with PX4 Firmware flashed on to it via QGroundControl.

**Frame:-** S550 Hex
**Gimbal:**- Tarot T-3D - III
**Flight Controller :-**PixHawk 2.4.8
**Transmitter**:-Flysky FS-I6S
**Receiver**:-Flysky FS-IA6B
**MAVLINK**:3DR
**OS**:Windows 10 Home
**Software QGroundControl** V 3.1.2

I would like to control the tilt of the gimbal (The Gimbal is connected to Channel1 AUX out on the Pixhawk) from the transmitter using its potentiometer on channel 6, could you advise the best way of doing this as I see no option to control the gimbal from QGroundControl under the camera settings.

Is it possible to set these as system parameters or is this functionality going to be included in the next release - I am happy to BETA test for you.

Kind Regards
An Enthusiast

DonLakeFlyer, 2017-01-31:
You should ask the firmware folks about this. I'm not sure if it is current supported by PX4 firmware.

closed 2017-01-31 by DonLakeFlyer

dagar, 2017-01-31:
@stefanoghazzali you can find gimbal documentation here - https://dev.px4.io/advanced-gimbal-control.html

stefanoghazzali, 2017-01-31:
Thanks Guys saw the Gimbal Control was not sure how to program the parameters into QgroundControl will give it a play and get back to you - Thanks in advance