mavlink/qgroundcontrol #4485: Same version param update should overwrite

jgoppert, 2017-01-29:
When developing firmware we are changing params a lot. It seems we have to delete ~.config/QGroundcontrol* each time so it will upload the new data. If the version numbers match it seems it won't update. The behavior  I think should be that if the uploaded version number is >= current it should update.

jgoppert, 2017-01-29:
I think the related logic is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/ParameterManager.cc#L1095

We maybe need to invalidate the cache if a firmware file is uploaded with a the same version as a file in cache.

DonLakeFlyer, 2017-01-31:
On the surface this seems like an ok change. Need to think a little more to make sure,

DonLakeFlyer, 2020-11-08:
Supported now. But will be replaced by COMPONENT_INFORMATION support

closed 2020-11-08 by DonLakeFlyer