mavlink/qgroundcontrol #4491: Segmentation fault (core dumped), Intel Aero

ChristophTobler, 2017-01-31:
I regularly get a ```Segmentation fault (core dumped), Intel Aero``` when I'm connect to the ```Intel Aero```.
Here's the output of the terminal
[image]
[image]

DonLakeFlyer, 2017-01-31:
If you aren't running either Master or very latest Stable first step is to run with that.

ChristophTobler, 2017-02-01:
@DonLakeFlyer Sorry for not mentioning... I'm running master that is maybe a week old. But I can try with the very latest

DonLakeFlyer, 2017-02-01:
That should be new enough.

DonLakeFlyer, 2017-02-02:
@ChristophTobler Do you have the ability to build QGC yourself? I don't have one of these boards so repro for me is tricky.

DonLakeFlyer, 2017-02-02:
Also can you run with --logging:full on the command line. Capture output and post. That may help.

Victry69, 2017-02-02:
ok @DonLakeFlyer  
 Hope this will help   If you need anything else  just let me know  .i'll do my best to find the logs in my Mac  ✌🏾

[log.txt](https://github.com/mavlink/qgroundcontrol/files/746798/log.txt)

![screen shot 2017-02-01 at 11 06 21 pm]

(https://cloud.githubusercontent.com/assets/6749950/22536826/128db948-e8d3-11e6-8c4c-0cfe25799bf4.png)

[Process:               QGroundControl [1296].pdf](https://github.com/mavlink/qgroundcontrol/files/746795/Process.QGroundControl.1296.pdf)

Victry69, 2017-02-02:
here's a different  format  ✌🏾 hope it helps 

[Date:Time:       2017-01-30 18:58:26 -0500.pdf](https://github.com/mavlink/qgroundcontrol/files/746809/Date.Time.2017-01-30.18.58.26.-0500.pdf)

ChristophTobler, 2017-02-02:
@DonLakeFlyer here's the output of QGC with ```logging:full``` http://pasted.co/eb3cbf4f

DonLakeFlyer, 2017-02-15:
@NaterGator @dagar Windows builds are still set up to store symbols and debug crashes isn't it?

dagar, 2017-02-15:
We're uploading symbols to s3, but I'm not sure what needs to be done with a windows symbol server to actually use them.

@NaterGator?

DonLakeFlyer, 2017-02-15:
Actually I have a stack trace included here and now that I look at that a stack trace on a different OS isn't going to help. It's deep in Qt Qml code which nevers tells you anything about what is really going on since it's all script processing.

Is there anyone from Intel that could lend/give me one of these boards to test with?

dagar, 2017-02-15:
@ChristophTobler how reproducible is this? Is it always happening at the end of param sync? Has it ever worked?

NaterGator, 2017-02-15:
Yes, windows builds are still storing debugging symbols. The only tricky thing is running pepto to act as a proxy, details are here: https://github.com/mavlink/qgroundcontrol/pull/2680#issuecomment-180029843

Obviously you also need a crash dmp file from a windows user.

DonLakeFlyer, 2017-05-25:
Removing this from gating 3.2 release. Can't do much without a board to test.

julianoes, 2024-10-15:
Just stumbled on this, clearly outdated, so closing.

closed 2024-10-15 by julianoes