mavlink/qgroundcontrol #4492: behaviour of calculate voltage divider

ChristophTobler, 2017-01-31:
I have encountered two weird behaviours:
* If you press ```calculate``` to calculate the voltage divider and then don't enter a number and click calculate you cannot enter a number afterwards. Instead you have to click ```close```, replace the ```--.--``` with the old value and then calculate again.
* If you click ```calculate``` (already in the sidebar) several times it keeps updating the value (increasing or decreasing, depending on your values).

DonLakeFlyer, 2017-02-01:
First problem is fixed. Still looking at second problem.

DonLakeFlyer, 2017-04-14:
> If you click calculate (already in the sidebar) several times it keeps updating the value (increasing or decreasing, depending on your values).

I tried this and it worked fine. The numbers move around a little, but average to around the same values.

closed 2017-04-14 by DonLakeFlyer