mavlink/qgroundcontrol #4494: auto file name based

Antiheavy, 2017-01-31:
Feature request:

When Qgc prompts to save Flight Data Log it would be neat to populate a default file name based on the system clock similar to this format:
QGCdatalog-YYYY-MM-DD-HH-MM-SS-xx.mavlink

This way a lazy user can just hit save (or Enter) and is less likely to hit cancel (..looks at self..).

[image]

dagar, 2017-01-31:
And optionally always save them. https://github.com/mavlink/qgroundcontrol/issues/3112

NaterGator, 2017-01-31:
Would be even more appropriate to try and scan the logfile for a valid time signature and use _that_ time to set the filename.

Antiheavy, 2017-01-31:
Optionally always save them would be best!  Although I would choose to start the log upon autopilot connection not just upon arm.  There is often lots of important pre-flight data recorded prior to arming when trying to troubleshoot a problem remotely.

closed 2017-03-11 by DonLakeFlyer